ros-controls/ros_controllers #363: No timeout in JointGroupVelocityController

AndyZe, 2018-10-15:
From testing in Gazebo, it seems that JointGroupVelocityController never stops when commands go stale. For example, say it received a velocity message 200 seconds ago. It will continue moving according to this last command, until something breaks/cannot move farther.

It seems like a timeout should be added to handle this case. Please inform me if a PR would be welcomed to add this. And also, how should the timeout period be established? I guess it should be one of the parameters in the controller.yaml file?

AndyZe, 2018-10-15:
Here's a video to show what I mean. I stop sending commands to the controller after a few seconds but the arm keeps drifting upward.

[https://drive.google.com/file/d/1vXsQ-oxcefxUcGOVdtnXHXSZ5CW9eB4g/view?usp=sharing](https://drive.google.com/file/d/1vXsQ-oxcefxUcGOVdtnXHXSZ5CW9eB4g/view?usp=sharing)

wxmerkt, 2021-11-09:
I agree with @AndyZe - we should have a time-out for velocity forward command controllers. I am happy to add it - would a PR be accepted?

Proposal to amend `velocity_controllers/JointGroupVelocityController`:
- Define the timeout (double, in seconds) in the forward_command_controller with a default of -1. If timeout < 0, it's not being applied (e.g. for position). 
- When timeout is a positive value, set the command to zero if no command has been received in the given timeframe. Zero is acceptable for velocity, but not for effort control (but that is a separate issue)

bmagyar, 2021-11-09:
Very welcome!