ros-controls/ros_controllers #369: JointTrajectoryController difference kinetic and melodic

agutenkunst, 2018-10-29:
https://github.com/ros-controls/ros_controllers/commit/e6bb3c6323a7ac1c842b43217bd8437b1f11f3bc broke the class interface due to the introduction of a new parameter to `updateTrajectoryCommand`
how about a cherry-pick on this?

Maybe https://github.com/ros-controls/ros_controllers/commit/5ff302aa742c992193fb0c463d1b59a903174aac from @ipa-mdl should also be taken? 

PR desired?

ipa-mdl, 2018-10-29:
For clarification: it broke the subclassing interface.
I think back-porting is the only way to make it compatible again.

> PR desired?

IMHO yes :)

agutenkunst, 2018-10-29:
Applying the same changes in kinetic-devel would work for me. How about that?

ipa-mdl, 2018-11-11:
Resolved in #370

closed 2018-11-11 by ipa-mdl