ros-controls/ros_controllers #382: Ability to reset odom in diff_drive_controller

aswinthomas, 2018-11-20:
While doing test runs in a real robot and in simulation, I would like to reset the environment to an initial state, once of which involves resetting odometry.

I tried stop and reload of the velocity_controller, but the object is still active in either in gazebo (for sim) and node that implements the controller_manager interface (on real robot). Hence the odom does not reset.

Would it be possible to provide a service in diff_drive_controller so that states such as x_ and y_ are reset?

bmagyar, 2018-11-20:
I think this would be useful. Thoughts?

@artivis @efernandez

fayyazpocker, 2020-11-19:
Any updates on this ?

alcantara09, 2021-05-12:
This [PR](https://github.com/ros-controls/ros_controllers/pull/564) addresses this issue.