ros-controls/ros_controllers #389: Joint trajectory controller does not build

wilcobonestroo, 2018-12-06:
In file included from [workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:75:0,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
[workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h: In function ‘Trajectory joint_trajectory_controller::initJointTrajectory(const JointTrajectory&, const ros::Time&, const joint_trajectory_controller::InitJointTrajectoryOptions<Trajectory>&)’:
[workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h:336:40: error: ‘to_string’ is not a member of ‘std’
       error_string = "Dropping all " + std::to_string(msg.points.size());
                                        ^
[workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h:336:40: note: suggested alternatives:
In file included from /usr/include/boost/exception_ptr.hpp:9:0,
                 from /usr/include/boost/thread/exceptional_ptr.hpp:10,
                 from /usr/include/boost/thread/future.hpp:23,
                 from /usr/include/boost/thread.hpp:24,
                 from /opt/ros/kinetic/include/actionlib/server/action_server.h:41,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:58,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
/usr/include/boost/exception/detail/exception_ptr.hpp:492:5: note:   ‘boost::to_string’
     to_string( exception_ptr const & p )
     ^
In file included from /usr/include/boost/dynamic_bitset.hpp:15:0,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:43,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
/usr/include/boost/dynamic_bitset/dynamic_bitset.hpp:1167:1: note:   ‘boost::to_string’
 to_string(const dynamic_bitset<Block, Allocator>& b, stringT& s)
 ^
In file included from [workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:75:0,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
[workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h:338:42: error: ‘to_string’ is not a member of ‘std’
       error_string += "Last point is " + std::to_string(last_point_dur.toSec());
                                          ^
[workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h:338:42: note: suggested alternatives:
In file included from /usr/include/boost/exception_ptr.hpp:9:0,
                 from /usr/include/boost/thread/exceptional_ptr.hpp:10,
                 from /usr/include/boost/thread/future.hpp:23,
                 from /usr/include/boost/thread.hpp:24,
                 from /opt/ros/kinetic/include/actionlib/server/action_server.h:41,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:58,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
/usr/include/boost/exception/detail/exception_ptr.hpp:492:5: note:   ‘boost::to_string’
     to_string( exception_ptr const & p )
     ^
In file included from /usr/include/boost/dynamic_bitset.hpp:15:0,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h:43,
                 from [workspace]/src/ros_controllers/joint_trajectory_controller/src/joint_trajectory_controller.cpp:34:
/usr/include/boost/dynamic_bitset/dynamic_bitset.hpp:1167:1: note:   ‘boost::to_string’
 to_string(const dynamic_bitset<Block, Allocator>& b, stringT& s)
 ^
ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/build.make:62: recipe for target 'ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/src/joint_trajectory_controller.cpp.o' failed
make[2]: *** [ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/src/joint_trajectory_controller.cpp.o] Error 1
CMakeFiles/Makefile2:8857: recipe for target 'ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/all' failed
make[1]: *** [ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j3 -l3" failed

wilcobonestroo, 2018-12-06:
Adding `add_compile_options(-std=c++11)` to the CMakeList.txt solves the issue.

ipa-mdl, 2018-12-06:
You are building the `melodic-devel` branch on kinetic, which is is not supported.
Please use the `kinetic-devel` branch or the binary release.

closed 2018-12-06 by ipa-mdl