ros-controls/ros_controllers #397: High jerks in acceleration profile on implementing joint_trajectory_controller: pos_vel_acc_controllers

prashanthitata, 2019-01-05:
I am using ros-kinetic version on Ubuntu 16.04 . I am sending trajectories from MoveIt to ros control to move my 7 DOF arm. To analyse trajectory, I compared trajectory given by MoveIt with the output of ros control and found that acceleration profile of ros control output has so many jerks and overshoots and is not following MoveIt trajectory.

I have subscribed from /position_trajectory_controller/command to get output from ros control. I have plotted the trajectory using octave in both cases.

I suspected that spline interpolation is not happening in ros control as my hardware interface is PositionJointInterface and controller I am using is

 position_controllers/JointTrajectoryController (while spline interpolation happens only when position, velocity , acceleration values are sent as inputs) .

I then changed my hardware interface to PosVelAccJointInterface and controller to pos_vel_acc_controllers/JointTrajectoryController  in relevant files(urdf,config and relevant cpp files) and found no changes in my acceleration profile.

PFA of acceleration profile in both cases
[image]

Can anyone please assist me on this??

 

Thanks and Regards,
Prashanthi Tata

prashanthitata, 2019-01-11:
hello... any update on  this??

7675t, 2019-01-14:
Hi, I think your problem description is not enough for the investigation. Could you show your problem as everyone can reproduce? 
BTW I introduce my [rqt_joint_trajectory_plot](https://github.com/tork-a/rqt_joint_trajectory_plot) tool to plot the joint profile :smiley_cat:

ipa-mdl, 2019-01-17:
Looks like your hardware cannot follow your command.
You might want to check your limits.
I'd guess that the error offset is quite high (in `state` topic).

I noticed that the `joint_limit_interface` (used in your HW), does not work properly in open-loop setups..

> I then changed my hardware interface to PosVelAccJointInterface [...] and found no changes in my acceleration profile.

It should not change; the generated trajectory is the same for all interfaces.

gavanderhoorn, 2019-01-17:
@ipa-mdl: could the `joint_trajectory_controller` be a weak point here?

I'm currently working with hw that absolutely requires jerk-limited trajectories and afaict the JTC is not able to guarantee that (quintic splines). It's all position controlled though, so iiuc open-loop with `ros_control` and the default controllers.

ipa-mdl, 2019-01-17:
>  could the joint_trajectory_controller be a weak point here?

Yes, it it not yet suitable for all control problems..
However, it might work better with sane input trajectories (something adapted to the HW capabilites).
MoveIt! trajectories are not that optimized for actual control.

gavanderhoorn, 2019-01-17:
> > could the joint_trajectory_controller be a weak point here?
> 
> Yes, it it not yet suitable for all control problems..

It is almost the go-to solution though if you'd want a relative straightforward-to-setup configuration with a `FollowJointTrajectory` interface and `ros_control`.

What would you recommend the OP uses / does?

prashanthitata, 2019-01-18:
> It is almost the go-to solution though if you'd want a relative straightforward-to-setup configuration with a FollowJointTrajectory interface and ros_control.

I ve tried to communicate using FollowJointTrajectory action too. Result is same with jerks and overshoots

bmagyar, 2019-01-18:
> I'm currently working with hw that absolutely requires jerk-limited trajectories and afaict the JTC is not able to guarantee that (quintic splines). 

Do you have a suggestion for a better interpolation? It shouldn't be hard to extend this part with other configurable interpolations.

prashanthitata, 2019-01-18:
> However, it might work better with sane input trajectories (something adapted to the HW capabilites).

> MoveIt! trajectories are not that optimized for actual control.

@ipa-mdl 
Yes... when i generated my own custom trajectory and sent, acceleration profiles of ros-control was following MoveIt! acceleration profile.
But the problem is, trajectory points to my 7 DOF arm are sent from moveit . So the high jerks are observed

bmagyar, 2019-01-18:
>But the problem is, in general,trajectory points to my 7 DOF arm are sent from moveit . So the high jerks are observed

Have you tried smoothing on the moveit side in a post-processing step?
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/planning_adapters/planning_adapters_tutorial.html#running-ompl-as-a-pre-processor-for-stomp

amjack0, 2021-01-03:
Not the solution, but one work around can be use of "Impedance Control" 
Acceleration = K_p(currentAngle - desiredAngle) + K_v (currentVelocity -  desiredVelocity) + desiredAcceleration
torque = M * Acceleration + Coriolis*Velcoity + Gravity + friction

You can use these computed torques and simulate your angle, velocity and acceleration waypoints on EffortController !