ros-controls/ros_controllers #404: Can't build ackermann_steering_controller on Raspbian

shvydky, 2019-03-04:
I am trying to install ROS on Raspbian Stretch Lite (Release date: 2018-11-13) from source. My installing script is the following:
```bash
sudo apt update && sudo apt upgrade
sudo apt-get install dirmngr
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
sudo apt update && sudo apt upgrade
sudo apt-get install -y python-rosdep python-rosinstall-generator python-wstool python-rosinstall build-essential cmake
sudo rosdep init
rosdep update

mkdir -p ~/ros_catkin_ws
cd ~/ros_catkin_ws
rosinstall_generator robot ros_control ros_controllers teleop_tools joystick_drivers rosserial nodelet serial xacro geneus --rosdistro kinetic --deps --wet-only --tar > kinetic-robot_ros.rosinstall
wstool init src kinetic-robot_ros.rosinstall

rosdep install -y --from-paths src --ignore-src --rosdistro kinetic -r --os=debian:stretch
sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/kinetic -j2
```
In 40-50 minutes building process fails with error in ackermann_steering_controller.cpp:

/home/pi/src/ros_controllers/ackermann_steering_controller/src/ackermann_steering_controller.cpp:460:41: error: invalid initialization of reference of type ‘const boost::shared_ptr<const urdf::Link> ’ from expression of type ‘urdf::LinkConstSharedPtr {aka std::shared_ptr<const urdf::Link>}’

Could anybody please advise me how to fix this error?

ipa-mdl, 2019-03-04:
You might need something like #246 applied to the code.

ipa-mdl, 2019-03-04:
Perhaps combined with 03ae571f5d1e905fc0a7f6e58a38d5ee63dd3013.

And according to [REP3](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021) Debian Stretch is not supported for kinetic. You could use Jessie image or try ROS melodic.

shvydky, 2019-03-19:
I have moved to Ubuntu Mate and the problems are gone.

closed 2019-03-19 by shvydky