ros-controls/ros_controllers #424: Use of derivative term for generating velocity command

adamheins, 2019-07-11:
It seems odd to me to use a derivative gain to generate a velocity command, instead of only using a feedforward term. When I have both, we have something like:

<a href="https://www.codecogs.com/eqnedit.php?latex=v=K_p(x_d-x)&plus;K_d(\dot{x}_d-\dot{x})&plus;\dot{x}_d" target="_blank"><img src="https://latex.codecogs.com/gif.latex?v=K_p(x_d-x)&plus;K_d(\dot{x}_d-\dot{x})&plus;\dot{x}_d" title="v=K_p(x_d-x)+K_d(\dot{x}_d-\dot{x})+\dot{x}_d" /></a>

which doesn't make sense to me, as we've got <a href="https://www.codecogs.com/eqnedit.php?latex=\dot{x}_d" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\dot{x}_d" title="\dot{x}_d" /></a> appearing twice. What I would expect is to omit the derivative term all together, i.e.

<a href="https://www.codecogs.com/eqnedit.php?latex=v=K_p(x_d-x)&plus;\dot{x}_d" target="_blank"><img src="https://latex.codecogs.com/gif.latex?v=K_p(x_d-x)&plus;\dot{x}_d" title="v=K_p(x_d-x)+\dot{x}_d" /></a>

Am I missing something?

miguelprada, 2019-07-11:
I agree. However, the same code (I assume you're talking about [this](https://github.com/ros-controls/ros_controllers/blob/melodic-devel/joint_trajectory_controller/include/joint_trajectory_controller/hardware_interface_adapter.h)) is used to generate effort commands, and in that case the derivative term may make more sense.

adamheins, 2019-07-11:
Ah okay this makes sense to me, so one can always choose to simply use a D gain of 0 for the velocity controller case.

Thanks for clearing this up!

closed 2019-07-11 by adamheins