ros-controls/ros_controllers #429: Check hw is_ready_for_cmds (or similar) in controllers?

gavanderhoorn, 2019-09-09:
First: I was not sure whether to post this here or on `ros-controls/ros_control`. Issues can be transferred now with GH, so is `ros_control` is more appropriate, please move the issue.

Second: this is more a discussion-type issue, not an actual problem that I'd like to see fixed or a bug report.

---

Without going into too much (implementation) detail for now: what are the thoughts of @ros-controls on whether controllers should be able to have more insight into the state of the hardware they are controlling?

As an example: the `joint_trajectory_controller` will only reject goals when it's not in the `running` state. It could be beneficial in some cases to also reject goals whenever the `hardware_interface` reports an error (example: hw for which some kind of system setup or initialisation is needed before attempting to execute any new trajectories).

By exposing the state of the hw (in some hw-agnostic and generic way) to the controller, it could reject goals with error messages such as:

> Can't accept new commands. Hardware reports it is not ready for new commands.

(ideally with a new and more descriptive `error_code` value in `control_msgs/FollowJointTrajectory` than `INVALID_GOAL`)

For controllers that don't accept goals (such as those only subscribing to `command` topics), they could ignore incoming commands.

This could be done by implementing a virtual resource that controllers could claim which encapsulates hw status and they could check periodically. By making it an optional resource, `hardware_interface`s that don't support the "hw status interface" would not get the benefit of the additional check.

gavanderhoorn, 2019-09-09:
An alternative could be for the program wrapping/hosting the `hardware_interface` and `controller_manager` to monitor the state of the `hardware_interface` and then to stop/start a (preconfigured) set of controllers.

This could even be done using a stand-alone node (that essentially maps something like `hw_state -> controller runtime state`).

As at least the management of controller execution status is a coordination level type activity, that could make sense.

It would not allow the controller to send back more appropriate rejection messages though.

bmagyar, 2019-09-11:
We had a similar experience when experimenting with mode switching motor modules. If the module takes more than one update loop to switch, it will currently get new commands which it won't be able to fulfill. 

Indeed this feels like something that could be handled at the interface level. Should this be the responsibility of each controller or be directly enforced by the controller manager?

gavanderhoorn, 2019-10-23:
The [controller_stopper](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/a7df18d3dbb570ec341bcd12b61f286fcd7fd244/controller_stopper) package/node part of the new UR driver would seem to be an example of the external coordination type approach I described above.