ros-controls/ros_controllers #431: vel_steering_offset might be wrong

facontidavide, 2019-10-16:
Hi, I was taking a look at this code:

https://github.com/ros-controls/ros_controllers/blob/melodic-devel/four_wheel_steering_controller/src/four_wheel_steering_controller.cpp#L386-L398

And I am puzzled by this line: 
``` c++
double vel_steering_offset = (curr_cmd_twist.ang*wheel_steering_y_offset_)/wheel_radius_;
```

I am quite positive that this is wrong, since you should use the velocity of the steering angle, instead of `curr_cmd_twist.ang`.

As a reference, the 4ws seems to be conceptually correct 

```c++
double angular_speed_cmd = curr_cmd_4ws.lin * (tan(curr_cmd_4ws.front_steering)-tan(curr_cmd_4ws.rear_steering))/wheel_base_;
double vel_steering_offset = (angular_speed_cmd*wheel_steering_y_offset_)/wheel_radius_;
```

Am I missing something?

bmagyar, 2019-10-18:
FYI @vincentrou

vincentrou, 2019-11-01:
I am not sure to understand your concern @facontidavide .
Both vel_steering_offset are based on current command  (one from the twist and the other from the 4ws).
By the way thanks for the refactoring.

facontidavide, 2019-11-02:
mathematically speaking what is the purpose of **vel_steering_offset**?

If it is to compensate for the rotation of the Z axis and the corresponding translation at the wheel level, the the former formula is wrong and the latter is correct.

Think about the case in which **curr_cmd_twist.ang** is a constant: **vel_steering_offset** should be probably proportional to the gradient of that value, not the value itself.

 > Both vel_steering_offset are based on current command

This doesn't mean it is correct.