ros-controls/ros_controllers #432: Does JointTrajectoryController assume that joints are rotational?

alain-m, 2019-10-17:
Does `JointTrajectoryController` assume that joints are rotational - continuous, revolute - instead of linear - prismatic -? I cannot find any such assumptions documented on the wiki or in the code, but maybe I missed something.
I see the [`angle_wraparound_` member](https://github.com/ros-controls/ros_controllers/blob/melodic-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller.h#L182), apparently used to differentiate continuous from revolute joints.
The calculation for the `state_joint_error_` on position at https://github.com/ros-controls/ros_controllers/blob/melodic-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L390 and that for the `state_error` on position at https://github.com/ros-controls/ros_controllers/blob/melodic-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L394 use [`angles::shortest_angular_distance`](http://docs.ros.org/indigo/api/angles/html/namespaceangles.html#a83debbedd331511aa914f393db576818) which will always give you a result in [-pi;pi].
It would be confusing to find that the position error on my linear joint mysteriously doesn't want to go beyond [-3.14meters;3.14meters], although a linear joint with such a length is probably not very common...

bmagyar, 2020-01-10:
I think you are quite right about this. Did you come around to solve it in the end or still building your 3+ meters long linear joint? :)

alain-m, 2020-01-10:
I don't have a 3+m long linear joint so I was never really affected by this, just wanted to point it out for the next guy :)

steinmn, 2022-03-25:
As "the next guy", thank you so much for this comment @alain-m. This was really giving me a headache until I found this!