ros-controls/ros_controllers #440: Remove rqt_joint_trajectory_controller dependency

matthew-reynolds, 2019-12-20:
For consideration:

The `ros_control` metapackage does not depend on `rqt_controller_manager`, so installing `ros_control` does not pull in any GUI dependencies.

In contrast, the `ros_controllers` metapackage _does_ depend on `rqt_joint_trajectory_controller`, and so it _does_ pull in GUI dependencies via `rqt`.

Thoughts on removing this dependency from the metapackage?

bmagyar, 2019-12-21:
I'm not opposed to this idea

matthew-reynolds, 2019-12-21:
I did a little bit of looking into how breaking this change would be, and whether it's realistic to target Kinetic/Melodic or if we should consider only Noetic.

From a quick [GitHub search](https://github.com/search?l=&p=2&q=rqt_joint_trajectory_controller+extension%3Alaunch&ref=advsearch&type=Code&utf8=%E2%9C%93), I just looked at repos that use `rqt_joint_trajectry_controller` in a launch file. Only found 48 usages, several of which are from the same repos. Many of those repos are small (1 contributor, <5 stars, few commits). I checked to see how they're installing the `rqt_joint_trajectory_controller`,  most of them are actually not even depending on it at all through their package.xml. A few provide a list of packages to install via apt.

Of course, this is only the stuff on GitHub, but it looks like there are few people using it in properly setup package, the majority of the use is just as a command-line debugging tool. It should not be too breaking of a change to pull this out of `ros_controllers`.

bmagyar, 2020-01-11:
`catkin_lint` will ask maintainers to add a direct dependency on `rqt_joint_trajectory_controller`, not `ros_controllers`. The metapackage has always served as a convenience installer, nothing more really.

matthew-reynolds, 2020-01-12:
PR #443 opened

PR #443 (2020-01-26): Remove rqt_joint_trajectory_controller from metapackage
Closes #440
closed 2020-01-26 by bmagyar