ros-controls/ros_controllers #453: Separate tests into dedicated packages

matthew-reynolds, 2020-02-05:
I'm proposing splitting out our tests into dedicated packages. These packages will live in the repo, but will not be part of the metapackage, and will not be released. This is similar to [`ros_comm`](https://github.com/ros/ros_comm).

The rationale for this change is to pull out the heavy test dependencies from the metapackage. In particular, the metapackage currently transitively depends on `gazebo_ros`, since it is used by tests in `ackerman_steering_controller`, `diff_drive_controller`, and `four_wheel_steering_controller`.

Would appreciate feedback before I open a PR. Also curious on your thoughts on naming of the test packages (In `ros_control`, our test packages are suffixed with `_tests` but were originally designed with a different intention. In `ros_comm`, they are prefixed `test_`).

bmagyar, 2020-02-07:
I am happy with either approach, would probably be better to follow `ros_comm`s example. What's your take on this @ipa-mdl?

ipa-mdl, 2020-02-20:
IMHO tests belong with the tested code.

The existing test packages have been created to not expose the test plugins.

matthew-reynolds, 2020-03-06:
> IMHO tests belong with the tested code

I would agree, but I think the cost of pulling in all these heavy dependencies outweighs the preference of putting the tests alongside the tested code.

> But we could still release them

Good point.

> The released metapackage does not depend on `gazebo_ros`

This is not true - On Melodic, `rosinstall_generator --deps ros_controllers` lists `gazebo_dev`, `gazebo_msgs`, and `gazebo_ros` as recursive dependencies, since they are required by `ackermann_steering_controller`. These can also be seen using `rospack` or `rqt_dep`.