ros-controls/ros_controllers #456: Upcoming Melodic release

bmagyar, 2020-02-12:
Holding until this list is merged:
* https://github.com/ros-controls/ros_controllers/pull/451
* https://github.com/ros-controls/ros_controllers/pull/283
* https://github.com/ros-controls/ros_controllers/pull/452

@seanyen @matthew-reynolds @ipa-mdl please comment/contribute to getting them merged/suggest others

matthew-reynolds, 2020-02-12:
- WIP: ros-controls/ros_control#404 equivalent. Like #451, I just want to do a thorough pass of the deps to make sure we're not missing anything in here after the `ros_control` cleanup.

bmagyar, 2020-03-09:
The NaN check did not make the new release, thanks to everyone contributing!

https://github.com/ros/rosdistro/pull/24002

closed 2020-03-09 by bmagyar