ros-controls/ros_controllers #462: Input and output of read() and write() methods in hardware_interface

fjp, 2020-02-13:
In the [diffbot test](https://github.com/ros-controls/ros_controllers/blob/melodic-devel/diff_drive_controller/test/diffbot.h), I am confused about the definitions of [read()](https://github.com/ros-controls/ros_controllers/blob/884f4c2605151684bd61a78a5e7e9a0bf5b928cc/diff_drive_controller/test/diffbot.h#L83) and [wirte()](https://github.com/ros-controls/ros_controllers/blob/884f4c2605151684bd61a78a5e7e9a0bf5b928cc/diff_drive_controller/test/diffbot.h#L95) methods. In my opinion the content of the write method should be swaped with the content of the read method.
I think the write() method of a hardware_interface should write to the actuators and therefore fill the cmd_ member. And the same for the read method: The members vel_ and pos_ are filled in the read method because the hardware_interface reads the data from the sensors.

The read method of the hardware_interface in [this repo](https://github.com/eborghi10/my_ROS_mobile_robot/blob/e04acfd3e7eb4584ba0aab8a969a74d6c30eed34/my_robot_base/include/my_robot_hw_interface.h#L81) seems correct to me. 

Could someone please explain the correct or intended way of how to define the read and write methods?

matthew-reynolds, 2020-03-06:
You are correct, the `read()` should be reading joint positions/velocities/efforts/etc from the hardware, and `write()` should send commands to the hardware. Another quick sanity check is that we always do `read()`, `update()`, `write()`, then `sleep()`, in that order. Definitely doesn't make sense to operate on stale sensor data, `read()` should be reading _from_ the hardware at the beginning of the loop.

I think the tests in `ackermann_steering_bot.h` and `four_wheel_steering.h` are also backwards. Maybe `rrbot` too.

Please feel free to contribute a PR :)

---

I think the reason these are this way is because the `rrbot` tests do the kinematics calculations in `write()`, which gave the wrong impression when new tests were written. IMO these calculations belong in `read()`, but I think this is mostly a matter of opinion since these calculations don't perfectly fit into the model of the `RobotHW` read/update/write cycle anyways.

fjp, 2020-03-06:
Thanks @matthew-reynolds for the clarification. I'll try to find some time to swap the definitions of the read and write methods within the next couple of days.

fjp, 2020-03-18:
@matthew-reynolds I just found this [pull request](https://github.com/ros-controls/ros_controllers/pull/364) which is what I had in mind too. However, it didn't get merged because according to @ipa-mdl the implementation is correct the way it is. I still think the current logic is backwards and it needs to be switched but don't see myself in the position to decide this.

matthew-reynolds, 2020-03-18:
Looks like @ipa-mdl is of the opposite opinion with regards to the kinematics calculations being in `read` or `write`, which is fine, as I mentioned above it's mostly a matter of opinion where those calculations happen since it doesn't really fit the model anyways.

He did agree that printing out the command in `read` was odd. I would still be happy to see a PR moving that print, and taking a look at the other tests too. A quick scan of `ackermann_steering_bot.h` shows a couple of similar oddities - Printing positions in `write`, and printing commands in `read`. I didn't look further, but other tests may have the same thing.

fjp, 2020-03-27:
@matthew-reynolds I've just looked at [`ackermann_steering_bot.h`](https://github.com/ros-controls/ros_controllers/blob/62270677a23a0812c82f73d36328fb90a739c409/ackermann_steering_controller/test/common/include/ackermann_steering_bot.h#L78) and [`four_wheel_steering.h`](https://github.com/ros-controls/ros_controllers/blob/62270677a23a0812c82f73d36328fb90a739c409/four_wheel_steering_controller/test/src/four_wheel_steering.h#L62). As you've mentioned the `read` and `write` methods there are also backwards. 

To keep this in line with the updated documentation of `hardware_interface::RobotHW` (waiting in this [PR](https://github.com/ros-controls/ros_control/pull/433)) I can swap the implementations of `read` and `write` in these two headers too.

bmagyar, 2020-04-01:
Both those test bots originate from the diff drive controller's code and we decided to go with a copy instead creating a common dependency package. Please update those parts as well

fjp, 2020-04-01:
Ok, I'll take care of this. Before I change anything though, I would like to run the tests and see them passing. Just so that I can check my changes didn't break something. Currently, in the latest `melodic-devel` branch some tests fail as described in #476. 

I have to admit I don't know what the tests are doing, but in case changing `read` and `write` methods isn't covered by a test, I can start working on this before #476 got solved.

matthew-reynolds, 2020-04-02:
+1 for updating the others as well.

This repo has had flakey tests for a while, I'm sure @bmagyar is quite used to hitting "restart" on the CI ;) I haven't looked at #476 yet, but I would say don't let it hold you up, just re-run the tests a few times to make sure you haven't broken them.

fjp, 2020-04-02:
Ok, I'll soon start to update the others as well. 

So far the tests were failing in my Ubuntu Bionic virtualbox all the time and take a lot of time to finish but I'll try to rerun them until I get them passing - hopefully. I'll update #476 if there's something new.

fjp, 2020-04-08:
Short update on issue with the current state:

Implementations of `read` and `write` methods are swapped in [`diffbot.h`](https://github.com/ros-controls/ros_controllers/blob/melodic-devel/diff_drive_controller/test/diffbot.h), [`ackermann_steering_bot.h`](https://github.com/ros-controls/ros_controllers/blob/62270677a23a0812c82f73d36328fb90a739c409/ackermann_steering_controller/test/common/include/ackermann_steering_bot.h#L78) and [`four_wheel_steering.h`](https://github.com/ros-controls/ros_controllers/blob/62270677a23a0812c82f73d36328fb90a739c409/four_wheel_steering_controller/test/src/four_wheel_steering.h#L62). Now the `read` and `write` methods follow the intended use of [hardware_interface::RobotHW](https://github.com/ros-controls/ros_control/blob/melodic-devel/hardware_interface/include/hardware_interface/robot_hw.h), see its documentation for details and [this related PR](https://github.com/ros-controls/ros_control/pull/433).

`rrbot` seems to do it already as intended in its [`rrbot_hw_real.cpp`](https://github.com/carlosjoserg/rrbot/blob/1aa362695a4399a6f11c5d2f9bf0d0663ba7f507/rrbot_hw/src/rrbot_hw_real.cpp#L127) and [`default_rrbot_hw_sim.cpp`](https://github.com/carlosjoserg/rrbot/blob/1aa362695a4399a6f11c5d2f9bf0d0663ba7f507/rrbot_hw/src/default_rrbot_hw_sim.cpp#L260), so I think this issue can be closed.

matthew-reynolds, 2020-04-09:
Thanks so much @fjp, looks good to me now too. Let's leave this open just until we finish out the documentation updates as well.

 - [x] #474 Swap implementation in `diff_drive_controller`
 - [x] #478 Swap implementation in `ackermann_steering_controller` and `four_wheel_steering_controller`
 - [ ] ros-controls/ros_control#433 Update documentation to make preferred data direction clearer.

matthew-reynolds, 2021-05-19:
ros-controls/ros_control#433 was superseded by ros-controls/ros_control#457, which has been merged. So this issue is resolved😄 Thanks again @fjp

closed 2021-05-19 by matthew-reynolds