ros-controls/ros_controllers #464: Usage with a ackermann vehicle simulation

KaranManghi, 2020-02-18:
Hi,
   I have a simulation of an Ackermann car which I got from : https://github.com/trainman419/ackermann_vehicle-1.
My question is, how do I integrate this model with the ros_controller to get the odometry?
Thanks...

Best,
Karan Manghi

callen5914, 2020-03-14:
@KaranManghi very good question! As you can tell this package is super supported and they are quick to help. I too am in the same boat. We had one sim working where can drive it around in gazebo but no ODOM. I followed their Steer_Bot_Gazebo rabbit hole and created all kinds of extra work.

Guess what!

I doesn't work. I am beginning to to hate ROS with a passion! The people that dev for it don't properly update and document their work so people like poke at the same 4/5 lines of code and wastes days if not weeks trying to figure this BS out!

KaranManghi, 2020-03-14:
@callen5914  I came across this link : https://robotics.stackexchange.com/questions/4486/ackerman-steering-model where they have some kinematic equations for odometry. I am trying to figure out if they actually work with the ackermann vehicle accurately. It looks like these equations are for a rickshaw kind of a model but ackermann equations should be similar or same... Hope that helps.

matthew-reynolds, 2020-03-14:
@KaranManghi It looks like that package doesn't actually use `ros_controller`'s `ackermann_steering_controller`, they're performing all the wheel speed calculations themselves. This is why you aren't seeing odometry data published. If you use the `ackermann_steering_controller` instead, it should publish odometry as long as the wheel separation and radius are specified.

@callen5914 The `steer_bot_controller` is unmaintained and deprecated. The current `ackermann_steering_controller` is the continuation of that package (See #356). I don't know anything about the old `steer_bot_controller` other than it's unmaintained, so I can't offer much help there, but if you switch to the `ackermann_steering_controller` I'm sure we can help. Please try to be constructive and remember that the ROS ecosystem is a big collection of open-source packages maintained by lots of different individuals and groups. Indeed there are holes the documentation of some packages, if you run into trouble please open an issue so the appropriate parties can update things.

[ROS Answers](https://answers.ros.org/) is often a better place for getting help with specific problems you're facing with your implementation, while GitHub is often better for problems with the package themselves. You may be able to find more resources there as well.

callen5914, 2020-03-19:
@matthew-reynolds 

> but if you switch to the ackermann_steering_controller I'm sure we can help.

I've switched for sometime now... I finally have my URDF in a state where it doesn't fail and the launch to Gazebo seems to work. However, when I switch to wireframe view and I start publishing to the topic `cmd_vel` I can see the two joints moving but the virtual parts do not. 

Isn't the rear wheels supposed to be controlled by the ackermann joint(rear_wheel)?

srmainwaring, 2020-03-26:
In response to a question elsewhere I've provided a couple of worked examples using the `ackermann_steering_controller` and `steer_bot_hardware_gazebo` plugin here: https://github.com/srmainwaring/steer_bot.

The models are based upon one of the test models provided in `steer_drive_ros` and have been updated for ROS Melodic.

The first example is a two wheeled vehicle that only needs the `ackermann_steering_controller` and works with the `gazebo_ros_control/DefaultRobotHWSim` in the `gazebo_ros_control` plugin .

The second example is a four wheeled vehicle that uses the `ackermann_steering_controller` in place of the `steer_drive_controller` and uses `steer_bot_hardware_gazebo` to provide the Gazebo / ROS integration. The usage is consistent with the documentation here: http://wiki.ros.org/steer_bot_hardware_gazebo?distro=indigo. For ROS Melodic you will need to use the patched version of `steer_drive_ros`:  `https://github.com/tsedl/steer_drive_ros.git`.

The one issue I've noticed is the front steering wheels appear to have the left / right angles transposed. This may be a problem with my URDF or with the Gazebo plugin.