ros-controls/ros_controllers #467: Upcoming Melodic release

bmagyar, 2020-03-09:
Wishlist:

* #283

 @ipa-mdl  @matthew-reynolds & whoever else is reading this, please feel free to add your preferred issues/PR-s to be addressed here and we will try to prioritize.

callen5914, 2020-03-14:
Yes please document implement much better. It's lousy!

bmagyar, 2020-03-14:
@callen5914 if there's a specific thing you have problems with we can try taking a look.

callen5914, 2020-03-17:
@bmagyar I have some issues with implementing the Ackermann Steering Controller.

To be a little more specific: 

1. The [Wiki](http://wiki.ros.org/ackermann_steering_controller) states you need a YAML file with a minimal setup as follows.

```yaml
mobile_base_controller:
  type: "ackermann_steering_controller/AckermannSteeringController"
  rear_wheel: 'rear_wheel_joint'
  front_steer: 'front_steer_joint'
  pose_covariance_diagonal: [0.001, 0.001, 1000000.0, 1000000.0, 1000000.0, 1000.0]
  twist_covariance_diagonal: [0.001, 0.001, 1000000.0, 1000000.0, 1000000.0, 1000.0]

```

So does it matter where this yaml file located. Do I need to copy and paste the header files and the cpp file to my package some where? On the wiki they say you can get the gazebo sim plugin from this really old cirkit-03. None of this makes any sense.

Cheers,

ros-gy, 2020-04-03:
If possible, it would be great to have some documentation on 4-wheel steering controller.
https://github.com/ros-controls/ros_controllers/blob/melodic-devel/four_wheel_steering_controller/src/four_wheel_steering_controller.cpp
Specifically, derivation or description of the equations for calculating wheel velocities/angles, and what the difference is between twist command block (line 375) vs 4ws command block (line 418).

matthew-reynolds, 2021-05-19:
@ros-gy I've split your request out to https://github.com/ros-controls/ros_controllers/issues/566.

I think this issue has outlived its usefulness so I will close it.

closed 2021-05-19 by matthew-reynolds