ros-controls/ros_controllers #473: Help Understanding Ackermann

callen5914, 2020-03-17:
Would anyone care to elaborate on what this means?

> a wheel joint frame center's vertical projection on the floor must lie within the contact patch

I have the four virtual wheels and two virtual steers and then an addition two links resembling a bicycle for the urdf file but still can't get this thing working.

```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="ursa_bot">

    <!-- Degree-to-radian conversions -->
    <xacro:property name="degrees_45" value="0.785398163"/>
    <xacro:property name="degrees_90" value="1.57079633"/>
    <xacro:property name="degrees_180" value="3.14159265"/>

    <!-- Inertia Generator -->
    <xacro:macro name="default_inertial" params="mass">
        <inertial>
            <mass value="${mass}" />
            <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0" />
        </inertial>
    </xacro:macro>

    <!-- Basic Geometry of Body -->
    <xacro:property name="body_length" value="1.91" />
    <xacro:property name="body_width" value="0.680618" />
    <xacro:property name="body_height" value="1.065022" />
    <xacro:property name="body_mass" value="20.3"/>

    <!-- Front Wheel Properties -->
    <xacro:property name="front_wheel_width" value=".254" />
    <xacro:property name="front_wheel_diameter" value="0.605" />
    <xacro:property name="front_wheel_mass" value="20"/>

    <!-- Steering Link Properties -->
    <xacro:property name="steering_width" value="${front_wheel_width/4}" />
    <xacro:property name="steering_diameter" value="${front_wheel_diameter/5}" />
    <xacro:property name="steering_mass" value="2"/>

    <!-- Rear Wheel Properties -->
    <xacro:property name="rear_wheel_width" value="${front_wheel_width+steering_width}" />
    <xacro:property name="rear_wheel_diameter" value="0.635" />
    <xacro:property name="rear_wheel_mass" value="22.8"/>

    <!-- Base Link -->
    <link name="base_link">
        <visual>
            <geometry>
                <box size="${body_length} ${body_width} ${body_height}"/>
            </geometry>
            <material name="blue" />
        </visual>
        <collision>
            <geometry>
                <box size="${body_length} ${body_width} ${body_height}"/>
            </geometry>
        </collision>
        <xacro:default_inertial mass="${body_mass}"/>
    </link>

    <!-- Steering Macro -->
    <xacro:macro name="steering_link" params="prefix reflect">

        <link name="${prefix}_steering_link">
            <visual>
                <origin xyz="0.0 0.0 0.0" rpy="${degrees_90} 0.0 0.0"/>
                <geometry>
                    <cylinder radius="${steering_diameter}" length="${steering_width}"/>
                </geometry>
            </visual>
[... 146 lines cut ...]
            <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>

    <transmission name="rear_wheel_joint_trans">
        <type>transmission_interface/SimpleTransmission</type>
        <joint name="rear_wheel_joint">
            <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
        </joint>
        <actuator name="rear_wheel_joint_motor">
            <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>

    <!-- Gazebo plugin for ROS Control -->
    <gazebo>
        <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
            <!-- <robotNamespace>/ursa_bot</robotNamespace> -->
            <!-- <robotSimType>ackermann_steering_controller/AckermannSteeringController</robotSimType> -->
        </plugin>
    </gazebo>
</robot>
```

callen5914, 2020-03-17:
So I think I made a small advance.

```xml
<joint name="front_steer_joint" type="revolute">
        <parent link="base_link"/>
        <child link="front_steer"/>
        <axis xyz="0.0 0.0 0.1"/>
        <origin xyz="0.7 0.0 0.0" rpy="0.0 0.0 0.0"/>
        <limit lower="-0.50" upper="0.50" effort="1" velocity="3"/>
    </joint>

    <joint name="rear_wheel_joint" type="continuous">
        <parent link="base_link"/>
        <child link="rear_wheel"/>

        <axis xyz="0.0 0.1 0.0"/>
        <origin xyz="-0.70 0.0 0.0" rpy="1.57 0.0 0.0"/>
        <limit effort="1.0" velocity="3.0"/>
    </joint>

    <transmission name="front_steer_joint_trans">
        <type>transmission_interface/SimpleTransmission</type>
        <joint name="front_steer_joint">
            <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
        </joint>
        <actuator name="front_steer_joint_motor">
            <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>

    <transmission name="rear_wheel_joint_trans">
        <type>transmission_interface/SimpleTransmission</type>
        <joint name="rear_wheel_joint">
            <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
        </joint>
        <actuator name="rear_wheel_joint_motor">
            <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>
```

The joints require some values. So I set inertial, collision, visual and then solved that part. I can switch to wire frame mode in Gazebo and see the two links modeled as cylinders. However, the front cylinder is laid flat. I wonder if that has something to do with my original question...

Now consider my yaml file.

```yaml
 joint_state_controller:
  type: joint_state_controller/JointStateController
  publish_rate: 50

mobile_base_controller:
  type: "ackermann_steering_controller/AckermannSteeringController"
  rear_wheel: "rear_wheel_joint"
  front_steer: "front_steer_joint"

  virtual_rear_wheels: ["right_rear_axle", "left_rear_axle"]
  virtual_front_wheels: ["right_front_axle", "left_front_axle"]
  virtual_front_steers: ["right_steering_joint", "left_steering_joint"]

  pose_covariance_diagonal:
    [0.001, 0.001, 1000000.0, 1000000.0, 1000000.0, 1000.0]
  twist_covariance_diagonal:
    [0.001, 0.001, 1000000.0, 1000000.0, 1000000.0, 1000.0]

gazebo_ros_control/pid_gain:
  left_steering_joint: { p: 10000.0, i: 10.0, d: 0.5 }
  right_steering_joint: { p: 10000.0, i: 10.0, d: 0.5 }
  right_front_axle: { p: 10000.0, i: 10.0, d: 0.5 }
  left_front_axle: { p: 10000.0, i: 10.0, d: 0.5 }
  right_rear_axle: { p: 10000.0, i: 10.0, d: 0.5 }
  left_rear_axle: { p: 10000.0, i: 10.0, d: 0.5 }
```

When launching I see new and improved output!

```
[INFO] [1584460826.044004, 0.543000]: Controller Spawner: Loaded controllers: joint_state_controller, mobile_base_controller
[INFO] [1584460826.052093, 0.548000]: Started controllers: joint_state_controller, mobile_base_controller
```

These topics are started which means I am starting to gain some traction with this controller I think:
`odom`
`mobile_base_controller`

When I run `rosrun rqt_gui rqt_gui` however, I get no movement on the the ackermann links or the virtual wheels and steers. I can see the topic odom.pose.x is changing even with the gazebo model not moving. Any help with this?

callen5914, 2020-03-19:
Joints are collapsing now.

Achllle, 2020-06-15:
I think you might get more help on this if you ask your question on answers.ros.org instead. The issue section is for problems with the code, and it seems like you're not reporting a bug here. 

When posting on ros answers, also include screenshots of the setup and any other information of what you're trying to do to make it easier for others to help you. You should also take look to see if that question has already been asked. I see a few similar questions pop up.

callen5914, 2020-08-01:
You actually wasted enough time to write this? After months of being a dead thread? Funny enough, I will send that right back at you by adding that if you looked through all those results, you would notice that I have indeed asked there.

bmagyar, 2020-08-02:
@callen5914 I think in a community which stays afloat because of individuals putting their own time into it or pushing their organizations for allowed open-source-time, it is important to be respectful.

I am now closing this thread, you should take @Achllle 's advice. My advice to you is to either work on your manners or hire a consultant who can solve these problems for you.

closed 2020-08-02 by bmagyar