ros-controls/ros_controllers #476: [ackermann_steering_controller] sporadic test failures

fjp, 2020-03-31:
Hi, when I run the tests for `ackermann_steering_controller` using `/ros_ws/src/ros_controllers/ackermann_steering_controller$ catkin run_tests --no-deps --this`, some tests fail. Then, when I run one individual test that failed, it sometimes passes but most often fails. Here is the output running `rostest --text ackermann_steering_controller.test`:

```bash
/ros_ws/src/ros_controllers/ackermann_steering_controller/test/ackermann_steering_controller_test$ rostest --text ackermann_steering_controller.test

...

[       OK ] AckermannSteeringControllerTest.testOdomFrame (2330 ms)
[----------] 3 tests from AckermannSteeringControllerTest (23503 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test case ran. (23503 ms total)
[  PASSED  ] 2 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] AckermannSteeringControllerTest.testTurn
```

Any idea why this is happening and is it related to #460? Do the tests pass on Travis? 

Some tests in `ackermann_steering_controller_limits_test` seem to fail too. Here is some output of `catkin run_tests --no-deps --this`:

```bash
/ros_ws/src/ros_controllers/ackermann_steering_controller/test/ackermann_steering_controller_test$ rostest --text ackermann_steering_controller.test

...

[ROSTEST]-----------------------------------------------------------------------
[ackermann_steering_controller.rosunit-ackermann_steering_controller_limits_test/testLinearJerkLimits][passed]
[ackermann_steering_controller.rosunit-ackermann_steering_controller_limits_test/testLinearAccelerationLimits][passed]
[ackermann_steering_controller.rosunit-ackermann_steering_controller_limits_test/testLinearVelocityLimits][passed]
[ackermann_steering_controller.rosunit-ackermann_steering_controller_limits_test/testAngularJerkLimits][passed]
[ackermann_steering_controller.rosunit-ackermann_steering_controller_limits_test/testAngularAccelerationLimits][FAILURE]
/home/fjp/git/ros_ws/src/ros_controllers/ackermann_steering_controller/test/ackermann_steering_controller_limits_test/ackermann_steering_controller_limits_test.cpp:189
Expected: (fabs(new_odom.twist.twist.angular.z - old_odom.twist.twist.angular.z)) < (0.25 + VELOCITY_TOLERANCE), actual: 1.04469 vs 0.27
--------------------------------------------------------------------------------

[ackermann_steering_controller.rosunit-ackermann_steering_controller_limits_test/testAngularAccelerationLimits][FAILURE]
/home/fjp/git/ros_ws/src/ros_controllers/ackermann_steering_controller/test/ackermann_steering_controller_limits_test/ackermann_steering_controller_limits_test.cpp:191
Expected: (fabs(new_odom.twist.twist.linear.x - old_odom.twist.twist.linear.x)) < (0.1 + VELOCITY_TOLERANCE), actual: 0.795466 vs 0.12
--------------------------------------------------------------------------------

[ackermann_steering_controller.rosunit-ackermann_steering_controller_limits_test/testAngularVelocityLimits][passed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 6
 * ERRORS: 0
 * FAILURES: 1
```

By the way, is is possible to provide a summary output log of `catkin run_tests --no-deps --this` for all tests? Or some other way to provide a summary?

Please let me know if you need more information on this issue or if there is something I can help with fixing this.

fjp, 2020-03-31:
Here's a test summary:

```bash
/ros_ws$ catkin_test_results build/ackermann_steering_controller/test_results/
ackermann_steering_controller/rosunit-ackermann_steering_controller_limits_test.xml: 12 tests, 0 errors, 2 failures, 0 skipped
ackermann_steering_controller/rosunit-ackermann_steering_controller_test.xml: 6 tests, 0 errors, 2 failures, 0 skipped
Summary: 74 tests, 0 errors, 4 failures, 0 skipped
```

bmagyar, 2020-04-01:
These tests were recently #463 stabilized, can you confirm you've got these changes?

fjp, 2020-04-01:
Yes, I've just cloned the `melodic-devel`  branch of `ros_controllers` repo and I can see the changes. After building the `ackdrmann_steering_controller` package and running the tests in a new catkin workspace, I can confirm that, despite the fix, some still fail unfortunately:

```bash
fjp@ubuntu:~/catkin_ws$ catkin_test_results build/ackermann_steering_controller/test_results/
ackermann_steering_controller/rosunit-ackermann_steering_controller_test.xml: 6 tests, 0 errors, 2 failures, 0 skipped
ackermann_steering_controller/rosunit-ackermann_steering_controller_wheel_radius_param_test.xml: 6 tests, 0 errors, 2 failures, 0 skipped
ackermann_steering_controller/rosunit-ackermann_steering_controller_wheel_separation_param_test.xml: 6 tests, 0 errors, 2 failures, 0 skipped
Summary: 74 tests, 0 errors, 6 failures, 0 skipped
```

Is Travis running those tests too? If so, where can we see the results? Thank you.

matthew-reynolds, 2020-04-02:
> Is Travis running those tests too? If so, where can we see the results? Thank you.

I haven't had a chance to dig into the rest of this issue yet, but to address this question, the Travis builds are linked in the README.md. [Here](https://travis-ci.org/github/ros-controls/ros_controllers) is the most recent build on Melodic, with the full log of job 1 [here](https://api.travis-ci.org/v3/job/669932697/log.txt). The tests you mentioned are indeed being run.

fjp, 2020-04-02:
Thank you for the links. 

As I've mentioned in #462, so far the tests are failing in my Ubuntu Bionic virtualbox all the time and take a lot of time to finish but I'll try to rerun them until I get them passing - hopefully. If this doesn't work, I'll post the output of the test result here after running `catkin run_tests --no-deps --this`.

bmagyar, 2020-04-03:
Oh! Running it on virtualbox & taking a ton of time is a clue that this is probably "normal". The fixes we have gotten for those tests make sure that with Travis' resources it works fine and is not flaky anymore. If you give it a more resource-constrained system, it may...

fjp, 2020-04-08:
I have rerun the tests of the `ackermann_steering_controller` inside virtualbox but the tests keep failing. As I haven't access to a more powerful machine at the moment, I think this issue can be closed. If it is of interest, I have attached the output of the tests:

[ackermann_test_output.txt](https://github.com/ros-controls/ros_controllers/files/4452549/ackermann_test_output.txt)

closed 2020-04-08 by fjp

bmagyar, 2020-04-09:
Thanks and let's reopen in case you see it in the future