ros-controls/ros_controllers #492: How to use ft_sensor plugin in ROS-Noetic?

Murtaza786-oss, 2020-05-27:
Hello I have a urdf with a ft_sensor plugin attached and which works fine in **ros-melodic with ubuntu 18.04**. The plugin in the urdf looks like,

```
      <!-- Gazebo FT sensor plugin -->
    
      <gazebo reference="wrist_3_joint">
        <provideFeedback>true</provideFeedback>
      </gazebo>
      <gazebo>
        <plugin name="ft_sensor_plugin" filename="libgazebo_ros_ft_sensor.so">
          <updateRate>200</updateRate>
          <topicName>ft_sensor/raw</topicName>
          <gaussianNoise>0.0</gaussianNoise>
          <jointName>wrist_3_joint</jointName>
        </plugin>
      </gazebo>
```

Now I have switched to **ROS - Noetic with Ubuntu 20.04**. I referred the path `/opt/ros/noetic/lib` where I found the file `libforce_torque_sensor_controller.so`. So I modified my plugin as,

```
      <!-- Gazebo FT sensor plugin -->
    
      <gazebo reference="wrist_3_joint">
        <provideFeedback>true</provideFeedback>
      </gazebo>
      <gazebo>
        <plugin name="ft_sensor_plugin" filename="libforce_torque_sensor_controller.so">
          <updateRate>200</updateRate>
          <topicName>ft_sensor/raw</topicName>
          <gaussianNoise>0.0</gaussianNoise>
          <jointName>wrist_3_joint</jointName>
        </plugin>
      </gazebo>

```
but after launch the launch file I cannot see the topic ft_sensor/raw being published. I have no idea what is the issue here. Please provide your suggestion here.

bmagyar, 2020-05-28:
You are trying to load a `ros_control` plugin as a gazebo plugin. 
What is the expectation here?

Here's an example of a bipedal robot using FT sensors *and* `ros_control` in gazebo: https://github.com/pal-robotics/reemc_robot/blob/kinetic-devel/reemc_description/gazebo/gazebo.urdf.xacro

Check where the gazebo plugin you were using is released from and see if it's available for Noetic yet.

closed 2020-05-28 by bmagyar

Murtaza786-oss, 2020-05-28:
Can you please tell me where can I check whether it is released for noetic or not?