ros-controls/ros_controllers #494: rqt_joint_trajectory_controller becomes freeze

lepalom, 2020-05-29:
I'm trying to make a pure Python 3 version of ROS and I'm checking ros_controllers. I'm using Python 3.7 in a debian buster distro with backported packages.

There's a strange behavior in rqt when I use rqt_joint_trajectory_controller. When I update the combo or init the controller, rqt freeze and no error message is showed. I'm a bit lost because I don't know what to check. It seems that the controller is still running and it's only a GUI problem.

I have realized that the code is a bit old but I have not found any other issue. Any idea?

bmagyar, 2020-06-02:
You may want to disclose which version of ROS/Ubuntu you are using, where did you install the software from (apt or from source) and perhaps how exactly you produced the error. Thank you

lepalom, 2020-06-02:
Maybe i was no enough clear. I'm using Debian Buster, with the Debian ROS pacakges in Sid backported. I'm using ros-control (0.18.1) build from source and repackaged. Also, ros-controllers is 0.16.1  build from source and repackaged.

I have patched all the shebangs to force Python 3 and more or less all is working. I have to admit that I don't like the patch. I have done another test and joint_trajector_controller seems working but when I run the graphical interface rqt, that one freeze.

I do not pretend that you reproduce the error. I just would like to receive any guidance because the same code was working with Python 2 but it failed with Python 3. And it failed in the graphical part, but I don't see what could fail.

I would try to upgrade the packages to Noetic to see what happens.

lepalom, 2020-06-03:
I think that I had some troubles with a mixed versions. I close the issue.

closed 2020-06-03 by lepalom

bmagyar, 2020-06-03:
If you can find Python3 issues on Noetic, please send us some PR-s fixing them. Earlier distributions may or may not support Python3 as the default Python version there is 2.7