ros-controls/ros_controllers #499: `enable_odom_tf` parameter in diff_drive_controller does not seem to be respected

thieraufc, 2020-07-22:
I have a simple differential driven robot with some sensor fusion, so I've disabled ros_control's default setting of publishing the odometry transform by setting `enable_odom_tf: false`. However, `rostopic info tf` shows that the robot_state_publisher is still publishing transforms, and rqt_tf_tree shows that the tf tree is alternating between two states as a result of the `odom` still being published as the parent link of the base_link.

In this image is the tf tree system as expected:
[image]

(as a quick note, the reason for having a `base_link` have a child of `origin_link` is that `origin_link` is a known point within my CAD model, but not an appropriate location for a base link, so a fixed transform is intentionally set here for convenience)

Here is the tf tree in its alternate state, which it switches to and from intermittently within about a second:
[image]

As you can see, this conflict splits the tf tree, preventing localization/mapping functionality. I believe what I'm looking at demonstrates that the `enable_odom_tf` parameter isn't being set as expected, because during startup I see the following snippet: 
```
[ INFO] [1595453550.479240400]: Odometry frame_id set to odom
[ INFO] [1595453550.481055135]: Publishing to tf is enabled
```

Please let me know if I can provide additional clarifying information or if there are things you recommend I try.

bmagyar, 2020-07-23:
The snippet below is correctly highlighting the issue you raised. You should check your configuration as the print you show above is right after grabbing the parameter from the param server:
```
    controller_nh.param("enable_odom_tf", enable_odom_tf_, enable_odom_tf_);
    ROS_INFO_STREAM_NAMED(name_, "Publishing to tf is " << (enable_odom_tf_?"enabled":"disabled"));
```

I do have a question though which is how `robot_state_publisher` comes into the picture here? The `diff_drive_controller` will publish to `/tf` directly, no middle man is used.

thieraufc, 2020-07-23:
That was helpful, thanks-- I've checked the rosparam server, and you were correct that the parameter is true when the node starts. I set the parameter in the launch file as shown in the test file, which set the parameter properly and solved the issue I was seeing. Upon checking the configuration again, I realized I had made a typo in spelling `enable_odom_tf`, so the parameter wasn't being set properly and that's my mistake.

Regarding robot_state_publisher: when the robot_state_publisher node isn't running, the tf data for my robot model isn't being published even though it is in the robot_description parameter. I'm now comparing my setup to the test setup to see if I've configured something improperly, preventing this from being published.

thieraufc, 2020-07-23:
Nothing is jumping out to me as something too odd with my setup, so I'm sure there's something I've missed... Do you have any tips as for what I might need to look at?

bmagyar, 2020-07-24:
The job of the robot state publisher is to publish `tf`. Static frames are acquired by parsing the robot description for the non-moving parts of your robot, moving parts will be updated using the corresponding values from the `/joint_states` topic.

matthew-reynolds, 2021-05-19:
I'm going to close this as the initial issue (`enable_odom_tf` not working) has been resolved. If you have specific questions or issues about the robot_state_publisher, feel free to continue the conversation here or in a new issue, or at https://github.com/ros/robot_state_publisher.

closed 2021-05-19 by matthew-reynolds