ros-controls/ros_controllers #500: diff_drive_controller fails to initialise and load

moshahin1, 2020-07-25:
Hi, I am trying to implement differential drive on my 4 Wheel robot similar to the Husky and Jackal however when I use the controller I get the following error:
Using Ubuntu 18.04, ROS Melodic 

```
[ERROR] [1595701441.454646278, 0.449000000]: Robot description couldn't be retrieved from param server.
[ERROR] [1595701441.454815515, 0.449000000]: Failed to initialize the controller
[ERROR] [1595701441.455088288, 0.450000000]: Initializing controller 'differential_drive' failed
[ERROR] [1595701442.457054, 1.426000]: Failed to load differential_drive
```

I have looked across the internet for the solution and made sure of the following:
1. Included the `gazebo_ros_control` gazebo plugin. 
2. Transmission includes` hardware_interface/VelocityJointInterface`
3. Joint names match the YAML `left_wheel` and `right_wheel` parameters  
4. Namespace in the YAML file matches the launch file 
5. The args in the launch contain the name of the controllers in the YAML file. 
6. Checked the package is installed (it is) :` diff_drive_controller /opt/ros/melodic/share/diff_drive_controller`

So what am I missing or doing wrong? The joint_state_controller loads but the differential drive fails to load. I have even followed the Jackal files to see where I went wrong but still no idea. If you would like to look at the YAML, URDF and launch file: https://answers.ros.org/question/357979/diff_drive_controller-failing-to-launch/  

I am new to ROS so I would like to thank you in advance for your help and apologies if this issue is on my behalf.

mateus-amarante, 2020-08-19:
Let me try to help you.

You are getting into this condition:
https://github.com/ros-controls/ros_controllers/blob/1fd8d25707e289700743925c93644b4295b1c486/diff_drive_controller/src/diff_drive_controller.cpp#L634-L638

Make sure you have your robot URDF model is loaded in "robot_description" param. The [urdf tutorials](http://wiki.ros.org/urdf/Tutorials) might help you with that

ingramator, 2020-08-25:
This is definitely a namespace issue. Unfortunately all throughout ROS these can sometimes crop up in ways that aren't intuitive. 

Do a `rosparam list` and check that your robot_description param is there and it's not under another namespace.

wxmerkt, 2020-08-25:
I believe one solution would be to change the instantiation to support non-root-namespace robot_description parameters. See #245 and #503 - this would enable namespace/multi-robot support.

moshahin1, 2020-08-25:
After messing around, I do think it is a namespace issue. The issue occurs when I insert `monsterborg:`(name of my robot) at the top of the YAML file with the`rosparams` for the `diff_controller`. I will also run the `rosparam list` command and check my `robot_description` is there. Would it be helpful to link the github repository with the code? I evantually want to spawn multiple robots so I want to use a different namespace for each one e.g. `monsterborg1/` etc...  

@wxmerkt I will look at those today and post the outcome ASAP. Thank you everyone!

closed 2020-10-10 by bmagyar