ros-controls/ros_controllers #501: "file name too long" issue during compilation in Windows

sujoykroy, 2020-08-02:
Hello,

I am having issue to build the package in windows. I am getting **file name too long** error.
Is there any command line switch in `catkin_make` to bypass this issue?

Steps to reproduce the issue:
On a Windows machine, install ROS noetic as per the instructions stated on http://wiki.ros.org/Installation/Windows.

Open a ROS environment with command like ,
`C:\Windows\System32\cmd.exe /k "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\VsDevCmd.bat" -arch=amd64 -host_arch=amd64&& set ChocolateyInstall=c:\opt\chocolatey&& c:\opt\ros\noetic\x64\setup.bat`

Assuming that ROS and its related files are in C:\opt\ros\noetic folder, run the following commands in that ROS environment shell.

`mkdir C:\opt\ros\noetic\workspace`
`mkdir C:\opt\ros\noetic\workspace\src`
`git clone https://github.com/ros-controls/ros_controllers`
`git clone https://github.com/ros-drivers/four_wheel_steering_msgs`
`git clone https://github.com/ros-controls/urdf_geometry_parser`
`catkin_init_workspace`
`cd ..`
`catkin_make -DCATKIN_ENABLE_TESTING=0`

Near the end of the compilation/building, following message appears,
[ 74%] Built target position_controllers
Scanning dependencies of target rqt_joint_trajectory_controller_rqt_joint_trajectory_controller_exec_install_python
[ 76%] Building CXX object ros_controllers/rqt_joint_trajectory_controller/CMakeFiles/rqt_joint_trajectory_controller_rqt_joint_trajectory_controller_exec_install_python.dir/catkin_generated/add_python_executable/rqt_joint_trajectory_controller_rqt_joint_trajectory_controller_exec_install_python/rqt_joint_trajectory_controller.cpp.obj
rqt_joint_trajectory_controller.cpp
**c1xx: fatal error C1081: 'CMakeFiles\rqt_joint_trajectory_controller_rqt_joint_trajectory_controller_exec_install_python.dir\catkin_generated\add_python_executable\rqt_joint_trajectory_controller_rqt_joint_trajectory_controller_exec_install_python\rqt_joint_trajectory_controller.cpp.obj': file name too long**
NMAKE : fatal error U1077: 'C:\PROGRA~2\MICROS~1\2019\COMMUN~1\VC\Tools\MSVC\1426~1.288\bin\Hostx64\x64\cl.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed

bmagyar, 2020-08-02:
Perhaps in 2021 Windows will get a compiler that supports longer file names :D 

Jokes aside, I believe that @seanyen may be able to help you

sujoykroy, 2020-08-08:
> Perhaps in 2021 Windows will get a compiler that supports longer file names :D
> 
> Jokes aside, I believe that @seanyen may be able to help you

Thanks for the jokes. It triggered me to try a funny but  effective work around. I just renamed the folder `/ros-controls/ros_controllers/rqt_joint_trajectory_controller` (of the local copy of `ros-controls` repo) into a shorter name like `rqt_jtc`. Then `catkin_make -DCATKIN_ENABLE_TESTING=0` worked. I am not sure of the repercussion of this rename operation. I hope the package will still be available as `rqt_joint_trajectory_controller`

sujoykroy, 2020-08-08:
@seanyen is there any way to configure the formation of names like *rqt_joint_trajectory_controller_rqt_joint_trajectory_controller_exec_install_python*. The patterns look like "<package_name>_<package_name>_exec_install_python". If I can change it into something shorter then the windows filename issue may get resolved without tweaking the folder name of the repo.

bmagyar, 2020-08-10:
Those generated names are internal to catkin AFAIK, you may want to dig there.

bmagyar, 2020-08-10:
There also seems to be a variable where you can configure it within Windows but never versions as this autodesk help ticket reveals: https://knowledge.autodesk.com/support/3ds-max/learn-explore/caas/sfdcarticles/sfdcarticles/The-Windows-10-default-path-length-limitation-MAX-PATH-is-256-characters.html

closed 2020-08-10 by bmagyar