ros-controls/ros_controllers #508: Installation of rqt_joint_trajectory_controller on Noetic gives Import Error: No module named rospkg.rospack

AnlageM, 2020-09-03:
Good morning, as in the title, the error I get in running is:

```
Traceback (most recent call last):  
File "/opt/ros/noetic/lib/rqt_joint_trajectory_controller/rqt_joint_trajectory_controller", line 5, in <module>
    from rqt_gui.main import Main
File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/main.py", line 40, in <module>
    from rospkg.rospack import RosPack
ImportError: No module named rospkg.rospack
```

I suppose the problem is that noetic targets only python3, so in my workspace I opened the script in the devel folder and add #!/usr/bin/env python3 and now it works. I am relatively new to git so I don't know whether I should suggest modifications here, please would you help me to do that? Thanks!

bmagyar, 2020-09-03:
I've pushed an update to noetic-devel. Could you build this repo locally and verify that it works for you please?

AnlageM, 2020-09-03:
I tried, using 
`sudo apt install ros-noetic-rqt-joint-trajectory-controller` 
and when opening the file in  /opt/ros/noetic/lib/rqt_joint_trajectory_controller I still see python and not python3

bmagyar, 2020-09-03:
Releases don't happen that quickly. I meant to clone this repo and build it in your workspace.

AnlageM, 2020-09-04:
I cloned the repo (tried either _rqt_joint_trajectory_controller_ or _ros_controllers_), built with `catkin_make_isolated`, and sourced my workspace, but ROS can't find the packages, even if the package is built (I can see the related folders in build and devel). The only way I can make ROS find the packages is installing by apt.

Sorry if I explained the whole procedure I am new of ROS and I want to ensure I am acting correctly.

PeterMitrano, 2020-10-06:
FWIW running `rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller` after `sudo apt install ros-noetic-rqt-joint-trajectory-controller` works correctly for me on 20.04 with noetic

bmagyar, 2020-10-10:
Thanks very much for confirming this @PeterMitrano , closing...

closed 2020-10-10 by bmagyar