ros-controls/ros_controllers #533: rqt_joint_trajectory_controller (and the entire hosting rqt GUI) crashes when "robot_description" param doesn't exist

130s, 2020-11-18:
I have these ROS params regarding Robot description.
```
$ rosparam list|grep -i desc
/ariac/arm1/robot_description
/ariac/arm2/robot_description
/sensors_description
```
When choosing the controller manager namespace from the pulldown, the entire rqt GUI that hosts `rqt_joint_trajectory_controller` crashes, not just the plugin.
```
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_joint_trajectory_controller/joint_trajectory_controller.py", line 274, in _on_cm_change
    self._update_jtc_list()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_joint_trajectory_controller/joint_trajectory_controller.py", line 246, in _update_jtc_list
    self._robot_joint_limits = get_joint_limits()  # Lazy evaluation
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_joint_trajectory_controller/joint_limits_urdf.py", line 21, in get_joint_limits
    description = rospy.get_param(key)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 467, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: 'robot_description'
```

bmagyar, 2020-12-01:
Just to make sure I understand the issue.

When there's no global `robot_description` specified nor the param name "remapping" is in effect with `robot_description:=/ariac/arm1/robot_description`, the node throws an ugly exception that makes it all the way to the top of the host app?

130s, 2020-12-02:
> When there's no global `robot_description` specified nor the param name "remapping" is in effect with `robot_description:=/ariac/arm1/robot_description`, the node throws an ugly exception that makes it all the way to the top of the host app?

- Not just the "ugly" exception but the process dies.
- If you mean "host app" as in the rqt's window, not just `rqt_joint_trajectory_controller` plugin view, then yes.