ros-controls/ros_controllers #537: New releases reverted on Noetic due to Debian builds failing for diff_drive_controller

bmagyar, 2020-12-01:
https://github.com/ros/rosdistro/issues/27401

Version 0.17...  build, 0.18... fails. 
Look for  `boost_headers` in the log to find the error
http://build.ros.org/view/Nbin_db_dB64/job/Nbin_db_dB64__diff_drive_controller__debian_buster_amd64__binary/

bmagyar, 2020-12-01:
@mateus-amarante @matthew-reynolds do you guys have any spare bandwidth to look into this?

mateus-amarante, 2020-12-01:
I'll take a look at it tomorrow.

matthew-reynolds, 2020-12-02:
Looks like the issue is the version of CMake.

The `FindBoost` macro uses `Boost::headers` for header-only libraries as of 3.15, but before that it was `Boost::boost`. Looks like Debian Buster uses CMake 3.13. (Out of interest, the required/recommended CMake versions in REP 0003 are a bit all over the place and don't give an easy version-proof solution to this).

The easy fix is to just go back to `find_package(Boost REQUIRED)`, rather than `find_package(Boost REQUIRED COMPONENTS headers)` or `find_package(Boost REQUIRED COMPONENTS boost)`. Anyways, that's how the examples in the FindBoost documentation handle header-only, and how we currently do it in `joint_trajectory_controller`, so should be all good.

closed 2020-12-02 by bmagyar

bmagyar, 2020-12-04:
Thanks for the quick assistance on this guys!

PR #538 (2020-12-02): Use version-agnostic FindBoost macro for header-only usage
Should fix #537, though I don't have a Debian machine to actually test this on. But these changes bring `diff_drive_controller` and `ackermann_steering_controller` in line with `joint_trajectory_controller`, which builds fine, so I think it's pretty safe.

See https://github.com/ros-controls/ros_controllers/issues/537#issuecomment-736919602 for more details.