ros-controls/ros_controllers #541: robot arm joint effort controller of openmanipulator-p (in real robot) can not work!

Medivh7, 2020-12-06:
Using moveit planning to control the robot arm to move. Robot arm joint effort controller of openmanipulator-p in gazebo can work well，but when i run in the real robot arm, it can't work!
when i run the joint effort controller, the real robot arm can output torque. i can push it by my hand, then it will return to the origin position (just like a spring). but when i run the moveit to control it, it will lock each joint, and can't move! 
we get the trajectory module warning:
[ WARN] [1607226630.131100399]: Controller arm_controller failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1607226630.131192346]: Controller handle arm_controller reports status ABORTED
[ INFO] [1607226630.131209539]: Completed trajectory execution with status ABORTED ...

destogl, 2021-01-08:
This is probably more appropriate question for people providing robot's driver.

matthew-reynolds, 2021-05-19:
Closing this as it's a problem in a downstream project or in the user's configuration, not in `ros_controllers`.

closed 2021-05-19 by matthew-reynolds