ros-controls/ros_controllers #545: Diff drive plugin only straight

grafoteka, 2020-12-22:
Hello, 

I have implemented the diff_drive plugin in a four wheels robot. 

When I simulate it in Gazebo, it moves correctly when I publish cmd_vel/lineal_x. Nevertheless, when I publish cmd_vel/ang_z the robot does not turn, but the topic is being published.

Does it require any special configuration?

Here I attach mine.

```
toa_joint_publisher:
  type: "joint_state_controller/JointStateController"
  publish_rate: 50

toa_velocity_controller:
  type: "diff_drive_controller/DiffDriveController"
  left_wheel: ['front_left_wheel', 'rear_left_wheel']
  right_wheel: ['front_right_wheel', 'rear_right_wheel']
  publish_rate: 50
  pose_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.03]
  twist_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.03]
  cmd_vel_timeout: 0.25
  velocity_rolling_window_size: 2

  # Base frame_id
  base_frame_id: base_link

  # Odometry fused with IMU is published by robot_localization, so
  # no need to publish a TF based on encoders alone.
  enable_odom_tf: false

  # Husky hardware provides wheel velocities
  estimate_velocity_from_position: false

  # Wheel separation and diameter. These are both optional.
  # diff_drive_controller will attempt to read either one or both from the
  # URDF if not specified as a parameter
  wheel_separation : 2.6
  wheel_radius : 0.3

  # Wheel separation and radius multipliers
  wheel_separation_multiplier: 1.0 # default: 1.0
  wheel_radius_multiplier    : 1.0 # default: 1.0

  # Velocity and acceleration limits
  # Whenever a min_* is unspecified, default to -max_*
  linear:
    x:
      has_velocity_limits    : true
      max_velocity           : 1.0   # m/s
      has_acceleration_limits: true
      max_acceleration       : 3.0   # m/s^2
  angular:
    z:
      has_velocity_limits    : false
      max_velocity           : 2.0   # rad/s
      has_acceleration_limits: false
      max_acceleration       : 6.0   # rad/s^2
```

Thanks in advance.

Kinds regards,
Jorge

matthew-reynolds, 2021-05-19:
> Does it require any special configuration?

It shouldn't, at first glance your yaml looks ok. Do your wheels have reasonable frictions in Gazebo? Have you tried plotting the actual wheel velocities to see if they're turning as expected (Rather than just looking at the whole robot)?

beeks2114, 2023-06-28:
Any Update on this issue? We are running into a presumably similar issue.

bmagyar, 2023-06-29:
Good point. I'll close this issue as @grafoteka hasn't provided any feedback since. We haven't heard any other complaints other than a few 10s of 1000s of robots using this controller just fine ;)

closed 2023-06-29 by bmagyar