ros-controls/ros_controllers #547: Flaky tests

bmagyar, 2021-01-01:
```
    [ackermann_steering_controller.rosunit-ackermann_steering_controller_wheel_radius_param_test/testTurn][passed]
    [ackermann_steering_controller.rosunit-ackermann_steering_controller_wheel_radius_param_test/testOdomFrame][passed]
    
    SUMMARY
    [NON-XML-CHAR-0x1B][1;31m * RESULT: FAIL[NON-XML-CHAR-0x1B][0m
     * TESTS: 3
     * ERRORS: 0
    [NON-XML-CHAR-0x1B][1;31m * FAILURES: 1[NON-XML-CHAR-0x1B][0m
    
    rostest log file is in /root/.ros/log/rostest-570fe3c46d99-23237.log
    -- run_tests.py: execute commands
      /usr/bin/python3 /opt/ros/noetic/share/rostest/cmake/../../../bin/rostest --pkgdir=/root/target_ws/src/ros_controllers/ackermann_steering_controller --package=ackermann_steering_controller --results-filename test_ackermann_steering_controller_radius_param_test_ackermann_steering_controller_radius_param.xml --results-base-dir "/root/target_ws/build/ackermann_steering_controller/test_results" /root/target_ws/src/ros_controllers/ackermann_steering_controller/test/ackermann_steering_controller_radius_param_test/ackermann_steering_controller_radius_param.test 
    -- run_tests.py: verify result "/root/target_ws/build/ackermann_steering_controller/test_results/ackermann_steering_controller/rostest-test_ackermann_steering_controller_radius_param_test_ackermann_steering_controller_radius_param.xml"
  >>>
build/ackermann_steering_controller/test_results/ackermann_steering_controller/rosunit-ackermann_steering_controller_wheel_radius_param_test.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- AckermannSteeringControllerTest testForward
  <<< failure message
    /root/target_ws/src/ros_controllers/ackermann_steering_controller/test/ackermann_steering_controller_test/ackermann_steering_controller_test.cpp:73
    The difference between fabs(new_odom.twist.twist.linear.x) and cmd_vel.linear.x is 0.02165762085097149, which exceeds EPS, where
    fabs(new_odom.twist.twist.linear.x) evaluates to 0.1216576208509715,
    cmd_vel.linear.x evaluates to 0.10000000000000001, and
    EPS evaluates to 0.01.
  >>>
```

@matthew-reynolds have you worked on a similar fix for the diff drive by any chance?

matthew-reynolds, 2021-01-01:
I ran into these flakey tests when testing the new CI, I've been poking around at it but don't have anything solid yet. I'll keep investigating it

matthew-reynolds, 2021-01-03:
Since we're adding this repo as a downstream test for many of our other repos, I'll put a bunch of work in this week to try to iron out these flakey tests. It will be a pain to have failures popping up all over the place due to a couple flakey tests in diff drive and ackermann drive controllers. Currently looking into resetting the odom accumulators as a potential source of some failures...

PR #550 (2021-01-14): Improve flakey tests
Closes #547, and should make it much less painful to merge all the new CIs that test downstream packages.

The flakey `diff_drive_controller` and `ackermann_steering_controller` tests seem to be because the accumulator in the odometry isn't being fully reset. The control loop runs at 100Hz, and the odometry accumulators have a window size of 10. So our 0.1sec delay isn't always sufficient to fully clear out the accumulator.

2.0sec (Used already in many of the tests, just not all) is plenty long enough to ensure the accumulators are fully reset. I did a `colcon test --rete [... cut]
closed 2021-01-14 by bmagyar