ros-controls/ros_controllers #554: Compliant controller and gravity compensation

arsh09, 2021-01-31:
Hello, 

This issue is not about a bug or problem. I am rather asking for a recommended/better way to write code. 

I implemented gravity compensation for a robotic arm using KDL but I mistakenly put it in HW interface class that I pass to controller manager and then I used switch methods from HW interface (canSwitch etc) to call these methods based on if the user load/unload from joint position trajectory to effort controller. 

The problem with this approach (as I can think of) is that it is not very robot-agnostic. Secondly, I noticed that joint_states topic frequency drops when I switched to effort controller (i.e. when gravity compensation functions are called within HW interface class). 

While reading a PR on compliant controllers, I realised that this is probably not the best way to achieve this. Can anyone point me in the direction as to how I should implement this controller so that the output of gravity compensation (i.e. efforts) are sent to effort controller. Should I re-write effort controller class and incorporate gravity compensation in it. If so, then if I want to implement impedance controller, would I have to do the same thing again? 

P.S: I am not sure if this is the right place to ask this. If not, I apologise and will close this issue. 

Thank you

piyu23srma, 2021-09-23:
Making your own controller is advisable. Good example are shown in [here](https://github.com/frankaemika/franka_ros/tree/develop/franka_example_controllers/include/franka_example_controllers). 

Better to implement KDL in effort controller.

arsh09, 2021-09-27:
Thank you @piyu23srma . 

I ended up using KDL and making a custom controller.

closed 2021-09-27 by arsh09