ros-controls/ros_controllers #556: Sub-millisecond /joint_states messages?

acrylic-origami, 2021-02-24:
I've noticed that the `/joint_states` publisher seems to be capped at 1000Hz (in simulation time): `publish_rate` higher than this don't change the actual frequency of the messages, while lower ones do of course. I'm less sure if the controllers themselves are able to accept messages at >1kHz simulation rate either, which is important to me as well.

These controllers are spawned with `controller_manager`, and are being simulated in Gazebo (ODE physics). Who's setting this cap: joint_state_controller, controller_manager, Gazebo, etc.? And is there any way around it?

---

Use case: I'm simulating some high-torque, low-speed servos on light links so the speed limiting is important to the sim accuracy. Right now, the servo travels its entire range within one `/joint_states` tick because it latches to the first effort message, which is stall torque. It also travels much (>50x) faster than its true max speed in this time, so I'd like to run the control loop more frequently to resolve the speed change and simulate the effort properly.

matthew-reynolds, 2021-02-24:
Neither `joint_state_controller` nor `controller_manager` have hard limits to their update rate, both should be freely configurable without issue. As you've used, the `publish_rate` param controls the `joint_state_controller`. Assuming you're using the `gazebo_ros_control` plugin, the `<controlPeriod>` sdf tag sets the controller manager update period if the tag exists, if not it defaults to the Gazebo physics engine step size.

So this sounds like a Gazebo limit to me - You can check `gazebo::physics::PhysicsEngine::GetMaxStep()` (eg. `model->GetWorld()->Physics()->GetMaxStepSize()`) to check the max time step that the physics solver will take, which I belive defaults to 0.001 sec.

According to http://gazebosim.org/tutorials?tut=physics_params&cat=physics, this seems to be reconfigurable in the `<physics>` tag of your world sdf.

You probably want to decrease that step size (And, if you're manually specifying your controller manager update period, update that to match. Or just leave it out so you match the physics engine step size.)

acrylic-origami, 2021-02-24:
@matthew-reynolds Ahh, so when Gazebo's max step size is set in the world SDF it expands the cap. We've been setting max step size in the GUI _after_ launch, so I guess `gazebo_ros_control` pulls the max step at launch time and holds onto it. Interestingly, the initial realtime update rate also seems to affect this, e.g. if initially too high, the publish rate is throttled as if it's hitting CPU limits, even when lowered in the GUI after launch.

Anyways, thanks for the detailed answer!

closed 2021-02-24 by acrylic-origami

matthew-reynolds, 2021-02-24:
> I guess `gazebo_ros_control` pulls the max step at launch time and holds onto it

Yep, that's exactly correct - the control period gets set when the plugin is loaded, and then isn't updated. The implementation is here, if you're curious: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/bb5c26c1a119996db0b9ad754cb6576933cfa49c/gazebo_ros_control/src/gazebo_ros_control_plugin.cpp#L138