ros-controls/ros_controllers #559: ROS control: rate.sleep() blocks in rostest

fjp, 2021-03-22:
Hi, I am using the `diff_drive_controller` from ROS control on a [real robot hardware](https://github.com/fjp/diffbot), which is working except that the robot starts to move uncontrolled at the beginning (probably due to division by zero time period), but this is another issue and probably unrelated to the following:

When I try to use the hardware interface and control loop in [my rostest](https://github.com/fjp/diffbot/tree/test-hardware_interface/diffbot_base/test) (based on the [tests in `diff_drive_controller`](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/diff_drive_controller/test/diff_drive_controller.test)), the execution seems to get stuck in the call to `rate.sleep()` at the end of the `while(ros::ok())` loop:

```cpp
#include <ros/ros.h>
#include <diffbot_base/diffbot_hw_interface.h>
#include <controller_manager/controller_manager.h>
 
int main(int argc, char **argv)
{
    // Initialize the ROS node
    ros::init(argc, argv, "diffbot_hw_interface");
    ros::NodeHandle nh;

    // Setup a separate thread that will be used to service ROS callbacks.
    // NOTE: We run the ROS loop in a separate thread as external calls such
    // as service callbacks to load controllers can block the (main) control loop
    ros::AsyncSpinner spinner(3);
    spinner.start();
    
    // Create an instance of your robot so that this instance knows about all 
    // the resources that are available.
    diffbot_base::DiffBotHWInterface diffBot(nh);
 
    // Create an instance of the controller manager and pass it the robot, 
    // so that it can handle its resources.
    controller_manager::ControllerManager cm(&diffBot);
    
    
    // Setup for the control loop.
    ros::Time prev_time = ros::Time::now();
    ros::Rate rate(10.0); // 10 Hz rate
    
    // Blocks until shutdown signal recieved
    while (ros::ok())
    {
        // Basic bookkeeping to get the system time in order to compute the control period.
        const ros::Time     time   = ros::Time::now();
        const ros::Duration period = time - prev_time;
        prev_time = time;
        
        // Execution of the actual control loop.
        diffBot.read(time, period);
        // If needed, its possible to define transmissions in software by calling the 
        // transmission_interface::ActuatorToJointPositionInterface::propagate()
        // after reading the joint states.
        cm.update(time, period);
        // In case of software transmissions, use 
        // transmission_interface::JointToActuatorEffortHandle::propagate()
        // to convert from the joint space to the actuator space.
        diffBot.write(time, period);

        //ros::spinOnce();
        
        // All these steps keep getting repeated with the specified rate.
        ROS_INFO_STREAM("BEFORE");
        rate.sleep();
        ROS_INFO_STREAM("AFTER");
    }
    return 0;
}
```

This loop is basically from the original ros controls documentation mentioned in the [ROSCon 2014 talk](https://vimeo.com/107507546) and [doxygen docs](http://docs.ros.org/en/noetic/api/hardware_interface/html/c++/).

In the above code, when running [the rostest](https://github.com/fjp/diffbot/blob/66d8df80ad57321b876543c6536cf1ed89430899/diffbot_base/test/diffbot_base.test#L1), I see the "BEFORE" output but not the "AFTER". Also the calls to `diffBot.read()` and `diffBot.write()` log their output only once.

What is the reason that this loop cycles continuously on the real robot hardware but blocks in the rostest?

Is this problem related to the timing issues mentioned in https://groups.google.com/g/ros-sig-robot-control/c/18VRoTEV1vM?pli=1 and https://github.com/PickNikRobotics/ros_control_boilerplate/issues/4 ? Should I work with `clock_gettime(CLOCK_MONOTONIC, &current_time);` or `std::chrono::system_clock::now();` instead of just `ros::Time::now()`? 

If so, what are the problems that might happen when using just `ros::Time::now()` like I have it done currently to compute the period?

In case the above code of the control loop has flaws I am willing to provide a PR to update the docs or suggest to link to a reference implementation, for example the [PickNikRobotics/ros_control_boilerplate](https://github.com/PickNikRobotics/ros_control_boilerplate) or some unit test?

fjp, 2021-03-22:
> What is the reason that this loop cycles continuously on the real robot hardware but blocks in the rostest?

Just found out that the blocking inside `rate.sleep()` happens because of `use_sim_time = true`, which is set in the [common test launch file](https://github.com/fjp/diffbot/blob/66d8df80ad57321b876543c6536cf1ed89430899/diffbot_base/test/diffbot_common.launch#L3).

When I don't set `use_sim_time`, then the loop iterates continuously, but I guess it would be better to work with the `/clock` topic in tests. Like it is done in your tests, e.g. here in [diffbot.cpp](https://github.com/ros-controls/ros_controllers/blob/a075bc36ebf5b81a7dafe911907cc1b781dd0185/diff_drive_controller/test/diffbot.cpp#L49).

Regarding my second question, I would still be interested to know, if using only `ros::Time::now()` should be avoided and an implementation similar to [PickNikRobotics/ros_control_boilerplate](https://github.com/PickNikRobotics/ros_control_boilerplate) is the recommended way? Thank you!

matthew-reynolds, 2021-03-22:
It's not clear to me why your `rate.sleep()` is sleeping forever, maybe something to do with the sim clock? Not really sure.

Regarding your other questions about clocks:
 - Generally you should use a monotonic clock source (eg. `ros::SteadyTime::now()`, `std::chrono::steady_clock::now()`, etc) for calculating the period passed to `ControllerManager::update()`. A realtime clock can be skewed or even jump in order to keep in sync with, for example, NTP servers. This can result in inaccurate or even negative periods, which cause a big mess in the controllers when they try to update their commands. By contrast, the monotonic clock will always be a steady, monotonically increasing source of time (Generally uptime of the system). This means the period will always be the actual duration between calls to update(), and thus the controller will respond as expected.
 - Generally you should use `ros::Time::now()` for calculating the timestamp passed to `ControllerManager::update()`. This is because some controllers (Including the diff_drive_controller, see [here](https://github.com/ros-controls/ros_controllers/blob/a075bc36ebf5b81a7dafe911907cc1b781dd0185/diff_drive_controller/src/diff_drive_controller.cpp#L552)) use `ros::Time::now()` internally to calculate things such as command timeouts. (IMO the controller should really take a stamped msg, so that the user can specify the time base they want so long as they're consistent in their commands and `ControllerManager::update()`, but that's beside from the point)
 - When it comes to actually running your read/update/write loop at a fixed rate, technically `ros_control` doesn't care as long as the `period` you pass in is an accurate reflection of the duration between calls to `ControllerManager::update()`. But you're probably better off using a monotonic source of time to avoid having time smears making your loop rate inconsistent.

matthew-reynolds, 2021-03-22:
Closing, feel free to reopen if you feel something is unresolved

closed 2021-03-22 by matthew-reynolds

fjp, 2021-03-22:
Thanks Matt this solves the issue for me. Having your explanation in the documentation would've been helpful though (at least for me). Not sure if this would be useful for others, as ROS2 is slowly taking over anyway.

matthew-reynolds, 2021-03-22:
Good idea. Melodic and Noetic will be around for quite a while yet, so if you're able to put up a PR it would be appreciated.

bmagyar, 2021-03-23:
One more general rule of thumb: whatever timing method you end up using, try to avoid mixing time from different sources.