ros-controls/ros_controllers #562: Rotation is discontinuous for skid-steer robot when using diff_driver_controller.

EpsAvlc, 2021-05-09:
Hello and appreciation for your work.
I use the diff_driver_controller for my own skid-steer robot.
When I send a twist of linear = 2.6, angular = 0.51 rad/s, the robot's rotation is wired.
[image]
However, when I use husky simulator and send the same twist, there is no problem.
[image]
Any suggestions? 
This is my config:
```yaml
mr1000_velocity_controller:
  type: "diff_drive_controller/DiffDriveController"
  left_wheel: ['front_left_wheel', 'rear_left_wheel']
  right_wheel: ['front_right_wheel', 'rear_right_wheel']
  publish_rate: 50
  pose_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.03]
  twist_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.03]
  cmd_vel_timeout: 0.25
  velocity_rolling_window_size: 2

  # Base frame_id
  base_frame_id: base_link

  # Odometry fused with IMU is published by robot_localization, so
  # no need to publish a TF based on encoders alone.
  enable_odom_tf: false

  # Husky hardware provides wheel velocities
  estimate_velocity_from_position: false

  # Wheel separation and radius multipliers
  wheel_separation_multiplier: 3 # default: 1.0
  wheel_radius_multiplier    : 1.0 # default: 1.0

  # Velocity and acceleration limits
  # Whenever a min_* is unspecified, default to -max_*
  linear:
    x:
      has_velocity_limits    : true
      max_velocity           : 1.0   # m/s
      has_acceleration_limits: true
      max_acceleration       : 3.0   # m/s^2
  angular:
    z:
      has_velocity_limits    : true
      max_velocity           : 2.0   # rad/s
      has_acceleration_limits: true
      max_acceleration       : 6.0   # rad/s^2

```

Hope for your reply!

matthew-reynolds, 2021-05-09:
At first glance, this looks like a simulation issue, rather than one with the diff drive controller. I would guess that the wheels are spinning correctly, but the robot is slipping. Have you verified that the friction of your wheels in Gazebo is reasonable and the simulation timestep is sufficiently small? I'm not a Gazebo expert, I can't really provide details on how to accomplish those, but they seem like good starting points.

To further verify that the diff drive controller is acting correctly, you could look at the `publish_cmd` topic to see what the controller is outputting to your motors. I think Gazebo also has a way to plot the joint's behaviour, which would be another good way to make sure the wheels are spinning as expected.

EpsAvlc, 2021-05-10:
> At first glance, this looks like a simulation issue, rather than one with the diff drive controller. I would guess that the wheels are spinning correctly, but the robot is slipping. Have you [...]

Your insight is right. I change the wheel link's friction parameters and it acts much better.

closed 2021-05-10 by EpsAvlc