ros-controls/ros_controllers #567: Controllers failing on using asyncExecute()

Shoriya, 2021-05-25:
Environment
ROS Distro: Melodic
OS Version: Ubuntu 18.04
catkin_make

Background
I am trying to excute a trajectroy while checking if the executing of the trajectory is completed or not. The initial some plans get executed without any error but it later crashes with an error as below

Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'joint6': expected: 0.00678015, current: 0.0257665

I am not sure what am I doing wrong.

Error
Controllers preemted and then aborted

I am attaching my test code...

static int execute_status;

void planning(moveit::planning_interface::MoveGroupInterface& group, ros::NodeHandle& node_handle)
{
const robot_state::JointModelGroup* joint_model_group = group.getCurrentState()->getJointModelGroup("manipulator");

moveit::planning_interface::MoveGroupInterface::Plan my_plan1;
moveit::planning_interface::MoveGroupInterface::Plan my_plan2;
moveit::planning_interface::MoveGroupInterface::Plan my_plan3;
moveit::planning_interface::MoveGroupInterface::Plan my_plan4;
moveit::planning_interface::MoveGroupInterface::Plan my_plan5;
moveit_msgs::MotionPlanRequest response;

moveit::core::RobotStatePtr current_state;
std::vector joint_group_positions;
bool success;

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////
current_state = group.getCurrentState();
current_state->copyJointGroupPositions(joint_model_group, joint_group_positions);

//Pick up loaction
geometry_msgs::Pose pickup_pose;
pickup_pose.orientation.w = 1;
pickup_pose.position.x = 0.198236761301;
pickup_pose.position.y = -0.113148163157;
pickup_pose.position.z = 0.0858897188011;

group.setPoseTarget(pickup_pose);

group.setStartStateToCurrentState();
group.setApproximateJointValueTarget(pickup_pose,"Link6");

success = (group.plan(my_plan1) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
group.asyncExecute(my_plan1);
while(execute_status != 3)
{
if(execute_status==3)
{ROS_INFO("execute %i",execute_status);}

}

ROS_INFO("execute status %i",execute_status);
ROS_INFO("%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%POSITION-1 DONE%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%");

current_state = group.getCurrentState();
current_state->copyJointGroupPositions(joint_model_group, joint_group_positions);

//Pick up loaction
pickup_pose.orientation.w = 1;
pickup_pose.position.x = 0.198231888501;
pickup_pose.position.y = 0.0937851806414;
pickup_pose.position.z = 0.08588867484;

group.setPoseTarget(pickup_pose);

group.setStartStateToCurrentState();
group.setApproximateJointValueTarget(pickup_pose,"Link6");

success = (group.plan(my_plan2) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
group.asyncExecute(my_plan2);
while(execute_status != 3)
{
if(execute_status==3)
{ROS_INFO("execute %i",execute_status);}

}
ROS_INFO("execute status %i",execute_status);
ROS_INFO("%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%POSITION-2 DONE%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%");

current_state = group.getCurrentState();
current_state->copyJointGroupPositions(joint_model_group, joint_group_positions);
group.setNamedTarget("home");

success = (group.plan(my_plan3) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
group.asyncExecute(my_plan3);
while(execute_status != 3)
{
if(execute_status==3)
{ROS_INFO("execute %i",execute_status);}

}
ROS_INFO("execute status %i",execute_status);
ROS_INFO("%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%POSITION-3 DONE%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%");

current_state = group.getCurrentState();
current_state->copyJointGroupPositions(joint_model_group, joint_group_positions);

//Pick up loaction

pickup_pose.orientation.w = 1;
pickup_pose.position.x = 0.198231888501;
pickup_pose.position.y = 0.0937851806414;
pickup_pose.position.z = 0.08588867484;

group.setPoseTarget(pickup_pose);

group.setStartStateToCurrentState();
group.setApproximateJointValueTarget(pickup_pose,"Link6");

success = (group.plan(my_plan4) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
group.asyncExecute(my_plan4);
while(execute_status != 3)
{
if(execute_status==3)
{ROS_INFO("execute %i",execute_status);}

}
ROS_INFO("execute status %i",execute_status);
ROS_INFO("%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%POSITION-4 DONE%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%");
current_state = group.getCurrentState();
current_state->copyJointGroupPositions(joint_model_group, joint_group_positions);
group.setNamedTarget("home");

success = (group.plan(my_plan5) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
group.asyncExecute(my_plan5);
while(execute_status != 3)
{
if(execute_status==3)
{ROS_INFO("execute %i",execute_status);}

}
ROS_INFO("execute status %i",execute_status);
ROS_INFO("%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%POSITION-3 DONE%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%");

}

void trajectory_result(const moveit_msgs::ExecuteTrajectoryActionResult::ConstPtr &msg)
{
execute_status=msg->status.status;
ROS_INFO("%i execute status",execute_status);
}

int main(int argc, char** argv)
{

ros::init(argc, argv, "testplanning");
ros::NodeHandle node_handle;
ros::AsyncSpinner spinner(0);
spinner.start();

moveit::planning_interface::MoveGroupInterface group("manipulator");

group.setEndEffectorLink("Link6");
group.setPoseReferenceFrame("mirobot_world");
group.setPlannerId("RRTConnectkConfigDefault");
group.setNumPlanningAttempts(10);
group.setPlanningTime(10.0);
group.allowReplanning(true);
group.setGoalJointTolerance(0.01);
group.setGoalPositionTolerance(0.01);
group.setGoalOrientationTolerance(0.1);

group.setNamedTarget("home");
group.move(); // Move to home target

sleep(2.0);

ros::Subscriber sub2 = node_handle.subscribe("/execute_trajectory/result", 1, &trajectory_result);
while(ros::ok())
{
planning(group, node_handle);
}

sleep(2.0);
ros::waitForShutdown();

return 0;
}

[image]

caioaamaral, 2021-05-25:
looks like joint6 is not on the start point (the one you've defined in the srdf with the moveit assistant gui)

matthew-reynolds, 2021-05-25:
I'm going to close this since it's a question about MoveIt, not ros_controllers. However, I would encourage you to post your question on https://answers.ros.org rather than as an GitHub issue on MoveIt's repo, as they prefer to keep their GitHub issues to actual issues with MoveIt rather than troubleshooting user setups.

A quick Google search points to some similar questions from the past: https://github.com/ros-planning/moveit/issues/694 and https://github.com/ros-industrial/stomp_ros/issues/14, for example. Perhaps that could be a useful starting point.

closed 2021-05-25 by matthew-reynolds