ros-controls/ros_controllers #584: [diff_drive_controller] requires controller_manager_msgs

acxz, 2022-01-31:
Obtaining the following error: 

```
-- Using CATKIN_DEVEL_PREFIX: /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-diff-drive-controller/src/build/devel
-- Using CMAKE_PREFIX_PATH: /home/acxz/vcs/git/github.gatech/ACDS/multirotor/multirotor_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/acxz/vcs/git/github.gatech/ACDS/multirotor/multirotor_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python (found suitable version "3.10.2", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Using empy: /usr/lib/python3.10/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-diff-drive-controller/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Using Python nosetests: /usr/bin/nosetests-3.10
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PythonInterp: /usr/bin/python (found version "3.10.2")
CMake Error at /opt/ros/noetic/share/controller_manager/cmake/controller_managerConfig.cmake:197 (find_package):
  Could not find a package configuration file provided by
  "controller_manager_msgs" with any of the following names:

    controller_manager_msgsConfig.cmake
    controller_manager_msgs-config.cmake

  Add the installation prefix of "controller_manager_msgs" to
  CMAKE_PREFIX_PATH or set "controller_manager_msgs_DIR" to a directory
  containing one of the above files.  If "controller_manager_msgs" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:68 (find_package)

-- Configuring incomplete, errors occurred!
```

og issue: https://github.com/ros-noetic-arch/ros-noetic-diff-drive-controller/issues/3

matthew-reynolds, 2022-01-31:
`controller_manager_msgs` is included transitively through `controller_manager`. Importantly, `controller_manager` is only used in tests, and the `find_package` is therefore wrapped in a `if (CATKIN_ENABLE_TESTING)` in the CMakeLists and in a `<test_depend>` in the package.xml.

See https://github.com/ros-controls/ros_controllers/blob/aa9c32b8ea8f83cb578cde82619f78e720fef89c/diff_drive_controller/CMakeLists.txt#L67 and https://github.com/ros-controls/ros_controllers/blob/aa9c32b8ea8f83cb578cde82619f78e720fef89c/diff_drive_controller/package.xml#L41

Sounds most likely to me that test dependencies are not installed - Perhaps the arch packaging isn't handle test dependencies?

acxz, 2022-01-31:
Looking into `controller_manager` I see the following:

https://github.com/ros-controls/ros_control/blob/53c2487d1b56a40f1d00b06c49512aa7fd2bf465/controller_manager/package.xml#L35

Yep looks like the arch packaging is not considering the `build_export_depend` tag properly.

I see. Is `CATKIN_ENABLE_TESTING` true by default?

matthew-reynolds, 2022-01-31:
> Is `CATKIN_ENABLE_TESTING` true by default?

I believe so. From [the Catkin docs](http://docs.ros.org/en/api/catkin/html/howto/format2/index.html#configuring-and-running-unit-tests), "All configuration steps related to testing should be only done conditionally when `CATKIN_ENABLE_TESTING` is set, **which is true by default.**" I don't think `catkin_tools` or `colcon` disables it, so I think it's default true with all build tools.

acxz, 2022-01-31:
Thanks for your help!

closed 2022-01-31 by acxz