ros-controls/ros_controllers #607: Disable cmd_vel_timeout

matthew-william-lock, 2022-09-26:
Hey !

Is there any way to disable the cmd_vel_timeout for the diff drive controller? It is not very apparent how to do this in the documentation.

See this [issue](https://github.com/cyberbotics/webots_ros2/issues/468) for more context.

bmagyar, 2022-11-18:
You want `ros_control` or `ros2_control`?  Any update on whether you guys managed to?

matthew-william-lock, 2023-02-24:
Looking for ```ros2```, no updates

bmagyar, 2023-02-26:
For that we have the ros2_controllers repo to open issues in.

First, let me show you the file where we define and describe the parameters:

https://github.com/ros-controls/ros2_controllers/blob/9c53e68c2bc4c35477f634c9e056e610da592d34/diff_drive_controller/src/diff_drive_controller_parameter.yaml#L77

bmagyar, 2023-02-26:
Reading this I can say there doesn't seem to be a way to disable it at the moment. You could consider adding a PR adding a Boolean enable_... parameter controlling it. If you wish to discuss this more please open an issue on ros2_controllers.

closed 2023-02-26 by bmagyar