ros-controls/ros_controllers #616: joint_state_controller will not publish when reset_controllers is held True

BryanStuurman, 2023-05-29:
I have a robot that, when it encounters an error condition (such as e-stop), it continually asserts reset_controllers=True on the controller manager update() method. This triggers a call to the joint state controllers starting() method which re-initializes the last_publish_time_ variable, effectively preventing the joint state controller update() method from ever publishing joint states.

I have built a quick fix to initialise last_publish_time_ once per joint state controller instantiation, found here:
https://github.com/ros-controls/ros_controllers/compare/noetic-devel...BryanStuurman:ros_controllers:fix-jsc-reset-publish

and I would be happy to submit a PR if the community so desires.

bmagyar, 2023-05-30:
> noetic-devel...BryanStuurman:ros_controllers:fix-jsc-reset-publish

Please do!

BryanStuurman, 2023-05-30:
aye aye sir!
https://github.com/ros-controls/ros_controllers/pull/617

BryanStuurman, 2023-12-06:
pr #617 merged, issue closed.