ros-controls/ros_controllers #621: How to user diff_drive_controller for a crawler chassis?

oscar1780, 2023-11-06:
[image]
   Hello everyone, I've got a crawler chassis as shown above, and i konw it's a diff driver mode. But i have no idea how to set wheel radius,if someone konw, please tell me. tks in adavance.

dcconner, 2023-11-06:
It should be the radius of your drive wheel to the tread surface.  Basically, the linear velocity of the tracks is v = w*r, where w is the angular velocity.  Presuming the black wheel in the right is your image is the drive wheel, then take the shaft to tread distance for radius.  This said, your question is better suited for https://robotics.stackexchange.com instead of this github.

oscar1780, 2023-11-06:
> It should be the radius of your drive wheel to the tread surface. Basically, the linear velocity of the tracks is v = w*r, where w is the angular velocity. Presuming the black wheel in the right is [...]

OK,thank for your reply,i'll post such question to this https://robotics.stackexchange.com website next time.

closed 2023-11-06 by oscar1780

bmagyar, 2023-11-06:
I'd set left and right wheels in the config file and play with the wheel radius multipliers to adjust them to reality. From that point on, the diff_drive_controller should work just fine