ros-controls/ros_controllers #626: JointTrajectoryController - spline interpolation strategy

fzoric8, 2023-12-17:
Hi, 

I'm wondering if it is possible to specify different interpolation method for the `JointTrajectoryControler`. 

Here in the **1.1. Trajectory representation** section, multiple  spline interpolation strategies are specified, but I'm not 
sure is it possible to change it from Quintic to Linear and how to do so. 
There's no parameter specification in the wki. 

As far as I'm able to see, Quintic interpolation is default in the `.cpp` source, but for my specific use-case I would
need `Linear` spline interpolation. 

So if there's a way to change spline interpolation strategy for the `joint_trajectory_controller` please let me know. 

Thanks!

dcconner, 2023-12-17:
If you only specify positions it does linear interpolation.  To do cubic you need to specify both position and velocity, and quintic needs position, velocity, and acceleration at each knot point.  It uses the appropriate interpolation based on the data provided.

fzoric8, 2023-12-17:
Nice, thanks for the quick answer!

closed 2023-12-17 by fzoric8