ros-controls/ros_controllers #628: Joint state controller throws exception when start time is less than update rate

01binary, 2023-12-23:
**The issue**

The joint state controller in `joint_state_controller.cpp` makes an assumption in its `JointStateController::starting(const ros::Time& time)` method that the time value passed to this function is always greater than its publish interval, which is set in Hz.

```
last_publish_time_ = time - ros::Duration(1.001/publish_rate_);
```

If the `time` passed into this function is less than `0.02` seconds for a `50` Hz publish rate, for example, this will result in `last_publish_time` being negative after the subtraction, which would throw a `std::runtime_error` exception causing the joint state controller to die:

```
terminate called after throwing an instance of 'std::runtime_error'
what():  Time is out of dual 32-bit range
```

The exception is thrown because the time in ROS is represented as an unsigned value, which would cause it wrap to the maximum possible value of the number type if a negative value was set.

*Proposed fix*

If we calculate the publish interval...

```
auto interval = ros::Duration(1.001/publish_rate_);
```

...and compare to `time` before subtracting...

```
if (time.toSec() > interval.toSec())
```

Then we can set the `last_publish_time` to `0` if the interval is larger than `time`, or set to their difference as before if the `time` is greater than `interval`.

*Testing*

I will submit a pull request for this issue, and test the joint state controller with Panda arm used in MoveIt tutorials ensure everything is working as expected.

Hopefully setting last publish time to zero will not break anything.

mjs973, 2023-12-23:
The joint_state_controller was broken by commit bba197e4 made on 2023-May-29.

01binary, 2023-12-23:
Sounds good, do you want `ros::Time::now()` or zero then?

mjs973, 2023-12-24:
@01binary Thank you for taking a look at this. It's unclear to me what commit bba197e was attempting to accomplish (since there's no discussion, and the commit message doesn't say.) 

In your patch, I don't see that it's useful to check for `last_publish_time_.isZero()` as this has a virtually zero chance of ever being true, even in simulation.

01binary, 2024-01-29:
Had a busy few weeks, sorry. So the failing test, as I mentioned is called `publishOK`. Here's how I run the test suite for joint state controller:

```
catkin_make run_tests_joint_state_controller
```
I get the following output:
```
[joint_state_controller.rosunit-joint_state_controller_test/publishOk][FAILURE]-
ros_controllers/joint_state_controller/test/joint_state_controller_test.cpp:140

The difference between real_pub_rate and pub_rate is 7.7832986269866815, which exceeds 0.05 * pub_rate, where
real_pub_rate evaluates to 17.783298626986682,

pub_rate evaluates to 10, and
0.05 * pub_rate evaluates to 0.5.
```

01binary, 2024-03-05:
This was addressed by #630 which was merged.

closed 2024-03-05 by 01binary